mavlink/qgroundcontrol #3601: Time and Speed slides in "Replay Flight Data" bar don't work

glody, 2016-06-20:
Test Environment
QGroundControl: Daily Build:v2.9.7b-282-g033f52b
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the time and speed slides in "Replay Flight Data" work well
Step:
1. Download tlog from here http://autotest.ardupilot.org/ArduCopter-test.tlog
2. Click "Replay Flight Data" on status bar
3. Select the tlog file to replay.
4. Try to change replay time and speed by dragging Time and Speed slides
Excepted result:
The replay time or replay speed should be change
Actual result:
The replay time or replay speed hasn't change

DonLakeFlyer, 2016-06-29:
This log is strange it's seems to have two vehicles in it: #255 and #1. How was it created? Should it have two vehicles?

DonLakeFlyer, 2016-07-01:
My testing on my own logs show that only the speed slider breaks things. So I removed it. Your log is showing up as multiple vehicles which is likely causing even more confusion with replay. That part I'm not planning on fixing.

DonLakeFlyer, 2016-07-07:
@glody If you can repro the mutiple vehicle during a log replay thing please enter a separate issue.

weakchicken, 2016-07-26:
QGC: daily build: HEAD:e6e0429 2016-07-20 18:58:20 -0400
Actually, the time slider can be moved, but it can't be changed into a precise position. When I drag the slider to a position and release it, it will flash to another side.

DonLakeFlyer, 2016-07-26:
Reopening for time slider problem.

DonLakeFlyer, 2017-05-17:
A bit better now with fixes. To change time, Pause and then change the slider, then Play. Speed changing still disabled since that was too big of a mess.

Making this even better will have to wait till 3.3

DonLakeFlyer, 2017-12-29:
Working now. Still need to pause to change time though.

closed 2017-12-29 by DonLakeFlyer