mavlink/qgroundcontrol #3602: Option "All" shouldn't parallel with other options in arm checking

glody, 2016-06-20:
Test Environment
QGroundControl: Daily Build:v2.9.7b-282-g033f52b
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the items in arm checking are reasonable
Step:
1. Launch QGC
2. Check the Arm checking list in Safety item, setting page
Excepted result:
Option "All" shouldn't parallel with other options, if "All" is checked, the other options should turn to gray.
Actual result:
Option "All" is paralleling with other options

DonLakeFlyer, 2016-06-24:
I'll look again. But I'm pretty sure that iven the way the parameter meta data works, there isn't way to make it work as you described.

DonLakeFlyer, 2016-06-24:
Yes the meta data for the param is a bit strange which causes this to happen. I could special case it, but it's a bit involved so pushing to 3.1

DonLakeFlyer, 2016-06-24:
Changed my mind. Playing it around with it some more, this is just too confusing the way it is.

DonLakeFlyer, 2016-06-25:
@glody Can you try this again with lastest daily QGC. Let me know if it now makes more sense that it used to.

closed 2016-06-29 by DonLakeFlyer

glody, 2016-07-04:
Checked on v2.9.7b-379-g2e9d112, the behavior is very excellent.

DonLakeFlyer, 2016-07-04:
Awesome, thanks for giving it a try.