mavlink/qgroundcontrol #3603: Sensors UI Screen

cfelipesouza, 2016-06-20:
Continuing the UI Reorg and due to Usability Tests, we are proposing some improvements on Sensors Screen.

Our idea, as you can see below, is to maintain the same structure of Summary Screen, adding some fields such as: a description of what that sensor does and bring some important information that helps the user to identify if the vehicle is working properly. 

We would like to have a list with all the sensors that could appear on screen, with a brief description. Details about each sensor could help us to put the values on each box. @dagar: could you help us on it?

[image]

Also, we added a title on the top of the page. The reason is to make clear to the user where he is.

Set Orientations panel will be placed inside each sensor calibration.

dagar, 2016-06-20:
This looks good.
What exactly are you looking for? A description of each sensor like internal magnetometer, optional external magnetometer, or higher level what each sensor type of sensor roughly does?
Are you looking for the parameters that correspond to each sensor?

For PX4 the possible sensors are

ACCEL0
ACCEL1
ACCEL2

GYRO0
GYRO1
GYRO2

MAG0
MAG1
MAG2

BARO

AIRSPEED

Most PX4 autopilots currently have 2 accels, 2 gyros, 1 internal mag, 1 baro. There's an optional external mag on the gps module (common for multicopters) and external airspeed sensor for planes.

biapalmeiro, 2016-06-21:
@dagar We are proposing to display a brief description for each sensor as a guidance for users. If it is possible and makes sense show the parameters for each of them also. So the user could have a more precise status of the items.

We would like to have this list of sensors, description and parameters, so we can see if the layout is working for all of them. And because it will be necessary for the implementation, to display the info it should be somewhere.

biapalmeiro, 2016-06-21:
@dogmaphobic Can you please label it as UX?

Thank you

DonLakeFlyer, 2016-06-21:
> Also, we added a title on the top of the page. The reason is to make clear to the user where he is.

I'm removing titles everywhere, including Settings and Setup. The reason being the button is selected, which I think add a decent amount of clarity. But the most important thing is that it takes too much important vertical real estate on small screens. It would be possible to have titles be optional and only show up on larger screens but I'm not sure that it's worth the difference. Is the missing title a big usability problem?

DonLakeFlyer, 2016-06-21:
Also there is a "Level Horizon" calibration that is in here as well. That doesn't have any information to show along with it.

DonLakeFlyer, 2016-06-21:
@LorenzMeier This new UI is great. But it's going to make the issue with devices not showing up until after you calibrate them even more confusing to the user. They'll go here and see that the stuff they have plugged in isn't showing up.

biapalmeiro, 2016-06-21:
@DonLakeFlyer Could you please explain better what you said to @LorenzMeier ? Just to see if we can think in a way to solve that. Thank you

DonLakeFlyer, 2016-06-21:
QGC uses specific parameter settings to determine which sensors are available and which are not. For example there is the possibilty of there being 3 compasses attached. QGC will adjust UI according to how many compasses are available. But the way the PX4 firmware works, QGC cannot determine the compasses which are connected until after you complete a calibration. Thus the initial screen that the user sees may end up being confusing since it doesn't show the hardware they have attached.

biapalmeiro, 2016-06-21:
Got it! Maybe for PX4 we don't show details for each sensor until the calibration is done.

DonLakeFlyer, 2016-06-21:
> Got it! Maybe for PX4 we don't show details for each sensor until the calibration is done.

That's the way the Vehicle Summary screen works now. It's just that having additional detail on the Sensors screen, the issue is more up front that it is now.

closed 2024-06-02 by HTRamsey