mavlink/qgroundcontrol #3604: No method to change default compass

glody, 2016-06-21:
Test Environment
QGroundControl: Daily Build:v2.9.7b-282-g033f52b
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the default compass is able to change
Step:
1. Launch QGC
2. Connect a real device which have two compass devices to QGC
3. Try to change the default compass
Excepted result:
User is able to select each one as the default compass.
Actual result:
No way to change the default compass. If the default compass doesn't work. The device will doesn't work even if the other compass is worked.

DonLakeFlyer, 2016-06-21:
If you are talking about the "Use Compass" setting from Mission Planner for example I just implemented that here: https://github.com/mavlink/qgroundcontrol/pull/3610. Is that what you mean?

glody, 2016-06-22:
Yes, that's what I mean. I have to use Mission Planner to switch Compass right now.

DonLakeFlyer, 2016-06-22:
Ok, this fixed now then.

closed 2016-06-22 by DonLakeFlyer