mavlink/qgroundcontrol #3608: Add compassmot support

DonLakeFlyer, 2016-06-21:


DonLakeFlyer, 2016-06-21:
http://ardupilot.org/copter/docs/common-compass-setup-advanced.html

davidbitton, 2016-06-21:
Is there a terminal mode in QGC where I could just call the command?

DonLakeFlyer, 2016-06-24:
@davidbitton I should have compassmot support done by tomorrow. When I put that in, would you be able to test it?

[image]
[image]
[image]

davidbitton, 2016-06-24:
Sure.

DonLakeFlyer, 2016-06-25:
@davidbitton Latest QGC daily has this in now. Can you give it a try?

davidbitton, 2016-06-25:
will do. i'm having an issue with calibrating my FVT 30A BLHeli ESCs.
Basically they won't. Baby steps.

davidbitton, 2016-06-26:
I keep getting "Check compass" and "Calibration command temporarily rejected". Could this be because I have only the external compass enabled? I cannot enable the internal or I receive inconsistent compasses.

DonLakeFlyer, 2016-06-26:
Possibly because compass mot is mostly used for internal compass. You'd have to ask the ardupilot firmware folks. Is this on a PixRacer? I haven't ever been able to get all compasses set up with ArduPilot. PX4 stack is happy with them though.

davidbitton, 2016-06-26:
Yes, it is in fact a PixRacer.

DonLakeFlyer, 2016-06-27:
I've never quite been able to get my PixRacer to be fully happy with ArduCopter. I fight with it every once in a while but then just don't have enough time to fight any more. Then I give up and go back to PX4 stack on it. And move back to my Iris for ArduCopter testing.

davidbitton, 2016-06-27:
are you compiling PX4 master?

DonLakeFlyer, 2016-06-28:
I normally use master when I'm test flying. Most of the time flashed with QGC. Sometime I'll build my own and flash from command line.

DonLakeFlyer, 2016-06-29:
Working now

closed 2016-06-29 by DonLakeFlyer