mavlink/qgroundcontrol #3611: Look into battery monitor calculator from MP

DonLakeFlyer, 2016-06-21:


DonLakeFlyer, 2016-06-23:
ArduPilot side completed. Need to look into same thing for PX4.

DonLakeFlyer, 2016-06-24:
@LorenzMeier As far as I can tell from looking at params it doesn't quite seem like the calculators are possible. I'm assuming the two params which need tweaking are BAT_A_PER_V and BAT_V_DIV. Let's use ArduPilot as example. It has a voltage divider parameter which always has a value in it. If the vehicle measured voltage is off you can update it using the formula:

(externally measured voltage * voltage divider) / vehicle reported voltage

But with BAT_V_DIV which I assume is the corresponding PX voltage divider param it defaults to -1. So there is no starting value to work from that you can adjust.

dagar, 2016-06-25:
In the code if the param is -1 the internal value is set based on hardware define.
https://github.com/PX4/Firmware/blob/571798c318a6736f748495378907cf76fae3c884/src/modules/sensors/sensors.cpp#L925

What if we change it to actually set the param?

DonLakeFlyer, 2016-06-25:
Yes, then the little calculator thingies would work. On the ArduPilot side Power settings has a set of built in settings which you can pick from a Combo. Then the Other selection shows you the dividers so you can tweak. Doesn't that sort of thing make sense with PX4 as well? You can run QGC + ArduCopter MockLink to see what it look like.

DonLakeFlyer, 2016-06-25:
Link to APM Power options: https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/APM/APMPowerComponent.qml#L54

dagar, 2016-06-26:
See https://github.com/PX4/Firmware/pull/4912

closed 2016-06-29 by DonLakeFlyer