mavlink/qgroundcontrol #3612: QGC logo in tool bar is not obvious to be a menu

AndreasAntener, 2016-06-22:
You don't necessarily click on the logo when looking for the settings menu, the actual menu icon in older versions was more clear (even if not so nice ;))

DonLakeFlyer, 2016-06-22:
I doubt we'll change this. @biapalmeiro - Does this show up in usability testing?

biapalmeiro, 2016-06-22:
We tested a previous version, so participants didn't deal with this. But we agree that the logo isn't obvious as the settings menu icon.

DonLakeFlyer, 2016-06-22:
I'd hate to lose the icon. We can fade it to the background a bit and put a hamburger over it.

DonLakeFlyer, 2016-06-22:
That said, this has been like this for quite some time now. And this is the first I've heard of anyone not being able to find settings. But I do get the issue.

DonLakeFlyer, 2016-06-22:
The one complaint I have heard is that people don't know what the big toolbar buttons do without clicking on them. I was thinking about maybe text over them at boot time that fades out to get out of the way.

biapalmeiro, 2016-06-22:
@DonLakeFlyer Can we keep it as it is for a while and we will think a better way to keep logo and menu icon on toolbar.

About the text over, it was one of the biggest concerns on the Usability Test. None of the components in the UI has tooltips that can guide users during their tasks.

DonLakeFlyer, 2016-06-22:
Tooltips don’t work since there is no such thing as a tooltip on a tablet.

NaterGator, 2016-06-22:
How about a persistent button that fades in tooltip style "what's this control?" overlayed text when pressed.

Works on mouse-based and touch devices alike.

pritamghanghas, 2016-06-22:
Most phone application solve this problem by showing first time animated overlays and text. That is kind of first time guide just explaining where is what. It gets out of the way after a few clicks on next/(ok got it).  I have seen that pattern in a lot of application especially google apps, it happens after every update.

biapalmeiro, 2016-06-23:
@DonLakeFlyer We have different behaviors on desktop and mobile, so the interface needs to adapt based on this. For people using on desktop, tooltips on mouse hover are very important. We need to find a way to solve this for mobile users.

DonLakeFlyer, 2016-06-23:
I would agree, but only if necessary. By that I mean, only if the mechanism for tablet simply will not work on desktop. The more the UI does things different on desktop versus tablet, the harder it is to support and maintain. In general the QGC UI has more of a tablet style UI with desktop being secondary. I'd like to try to find a tablet solution that will work on desktop as well.

AndreasAntener, 2016-06-23:
Why not just leave the logo there, and add the menu icon next to it? The logo could then open, ehm, an about dialog if anything.

But yes, it has been like this for a while, on the other hand I almost never used this menu in daily usage.

biapalmeiro, 2016-06-23:
We are working in a proposal for this issue, give us some days and we can discuss that. What do you think?

It is bigger than the logo x menu icon, there are some Information Architecture questions involved.

@anselmolsm @cfelipesouza

cfelipesouza, 2016-06-28:
We've doing some tests and here we have our proposal for the QGC header.

[image]

Some points:

1. We brought the purple to the entire bar, to align with the new QGC brand.

2. We put all the main menu together, as a clickable list. It's the same behaviour as vehicle name and flight mode. This way we have labels for the main actions and it's a way to divide the information.

3. The logo isn't clickable anymore. Settings (hamburger) was moved to the other side (three dots button). It's good a way to divide the drone settings against software settings.

4. All notifications will be found on the icon with the red dot. It can blink - if needed - and when you visualize the information the red dot disappear.

5. We are refining some icons, like satellite signal and telemetry, to align with the other icons already sent.

6. We are finishing the mobile proposal but we need to understand first if Vehicle Name is a item that needs to be always visible. Can you help us?

Waiting for comments.

DonLakeFlyer, 2016-06-28:
So far seems reasonable to me. Comments:
- The Vehicle name (Vehicle #129) only shows up if you have more than one vehicle connected. Normally (single vehicle) it would not be seen.
- I think I'd just put settings under the left hand drop down and call it something like App Settings. The right hand dots are a known thing on Android for a menu. But aren't known to iOS or desktop people.
- What does the airplane graphic between battery and vehicle id dropdown do?

I think we should consider this for 3.0 for a couple reasons:
- It a fairly significant change to the most prominent ui. So the sooner the better.
- The toolbar shows up in all screen shots. Don't want to reshoot every single screenshot for the user manual.

cfelipesouza, 2016-06-28:
> What does the airplane graphic between battery and vehicle id dropdown do?

We found this icon as telemetry. Does not exist?

cfelipesouza, 2016-06-28:
> The toolbar shows up in all screen shots. Don't want to reshoot every single screenshot for the user manual.

@biapalmeiro confirms that she can update the User Manual.

When will you release the 3.0 version?

dogmaphobic, 2016-06-28:
On the iPad, that will mean you have to click on the one and only button, wait for that weird "list box" to show up, scroll to the desired target function and hit "Done". It seems a lot of work to switch modes. Especially if the app always boots straight into "Vehicle Setup", which in theory, you would only have to deal with once in a blue moon.

DonLakeFlyer, 2016-06-29:
I wouldn't use a Menu since those are crop on iOS. Also we by default boot to Fly view unless Setup is incomplete. That said you are right about the change adding an extra click. I don't think you really move back and forth between views that much when you are really flying as opposed to testing so not sure if it's a big issue or not. That's kind go the key question to me.

biapalmeiro, 2016-06-29:
I really don't think people change views too often. When are you planning to release the 3.0? Does telemetry icon exist?

DonLakeFlyer, 2016-06-30:
This is what I was thinking: https://youtu.be/Ws8vkf3LlgQ. This is a video recorded from app boot. We can use the same toolbar buttons but still have a little visual style to the ui. Note: I haven't implemented help text on the Q button yet.

DonLakeFlyer, 2016-06-30:
Better one: https://youtu.be/pw7Fpbtk_N0

DonLakeFlyer, 2016-06-30:
Similar thing could be done with map round buttons

biapalmeiro, 2016-07-04:
@DonLakeFlyer I don't think it solved the initial problem. What do you think, @AndreasAntener? 

-It just appears once, so users need to remember icons' meaning;
-The line under the logo doesn't follow the branding guideline, it is important to keep the app's identity;
-It doesn't follow any platform user interaction standard;
-Not sure if users really need the tabs, so maybe we are focusing the energy on the wrong problem.

By the way, regarding the header discussion, we've collected some information about different patterns on different systems.

Each platform behaves differently. Applications should match the majority of patterns already established by each OS to merge with the environment and decrease the learning curve of users. Both Apple and Google discourage to make a literal port of an application for both systems.

**Actions**
Both systems use action buttons in different places. Android usually places actions on the bottom, while iOS places on top.

**Tabs**
It is also inverted in both systems. While Android keep tabs on top, Apple prefer to use it on the bottom.

[image]
iOS

[image]
Android

**Headers**
Due to the hardware form factor, applications on iOS must have a header able to hold a back button, while Android apps do not necessarily need, by the fact that the back button is a system action on Android. 

**Usage of icons X labels**

**Icon usability - Nilsen Norman Group**
“The usage of labels reinforce the meaning and reduce the ambiguity”
https://www.nngroup.com/articles/icon-usability/

**Do Icons Need Labels?**
“ A good user experience can be defined in many ways, but one measure is the ability to reduce the user’s effort to think.”
https://uxdesign.cc/do-icons-need-labels-6cb4f4282c00#.u1j819yaj

**Icons vs Lables vs Both**
“When words and icons are closely entwined, we augment intelligence by increasing ‘human bandwidth'—the capacity to take in, comprehend, and more efficiently synthesize large amounts of new information”.
http://edwardsanchez.me/blog/13589712

**Myth 13. Icons enhance usability**
“In most projects, icons are very difficult to get right and need a lot of testing. For abstract things, icons rarely work well”.
http://uxmyths.com/post/715009009/myth-icons-enhance-usability

**About number of clicks/taps**

**Seven Deadly Mobile Myths: Josh Clark Debunks the Desktop Paradigm and More.
MYTH #4: Extra taps and clicks are evil**
“In mobile, tap quality is far more important than tap quantity. As long as each tap delivers satisfaction, extra taps are good. Taps invite conversation—give and take—that you can get at and explore. Building meaningful click sequences are a form of progressive disclosure that helps you uncomplicate complexity”.
http://www.forbes.com/sites/anthonykosner/2012/05/03/seven-deadly-mobile-myths-josh-clark-debunks-the-desktop-paradigm-and-more/#2b81f3665270

**Satisficing**
“No one’s going to argue that we should add superfluous clicks in interfaces. But making “as few clicks as possible” an optimization goal is how interfaces become bloated and crammed with cruft”.
http://ignorethecode.net/blog/2012/12/17/satisficing/

**Designing for the Next Step**
“Almost every screen we design can be improved by really focusing on the steps and sequences of steps a user goes through. In our haste we often speed up the process too much, get steps out of order, fail to present an appropriate next step, or otherwise break the sequence. By re-assessing your app or site in light of these potential errors, you can discover the sequence and timing that your users need to successfully make it to the next step”.
http://bokardo.com/archives/designing-for-the-next-step/

@DonLakeFlyer @anselmolsm @cfelipesouza @fabricionovak

cfelipesouza, 2016-07-04:
I'm sending again our proposal with the reviewed icons and now with the menu opened as a reference. We endorse this idea because it divides the information between actions (menu x drone information) and it is possible to reduce for small screens with little changes.

[image]

[image]

DonLakeFlyer, 2016-07-04:
The line under the icons indicates selection. It’s not part of branding. 

I think users will pick up what the main toolbar icons do very quickly. If they don’t use the product for a while or it’s their first time, the text is meant to jog their memory. Not to mention other than the Q icon, the icons themselves convey the meaning. Also showing the icons at all times lets the user know immediately the full list of main capabilities of the application without having to dig around for it. Both single click access to the most used functions and show the user to top level functions is the kind of the whole idea for having a toolbar. 

> Each platform behaves differently. Applications should match the majority of patterns already established by each OS to merge with the environment and decrease the learning curve of users. Both Apple and Google discourage to make a literal port of an application for both systems.
We just don’t have the resources to implement or test an app which adjusts ui to each OS standard (screen size is hard enough to deal with). Nor do I think it is completely necessary. The question isn’t whether the ui follows a standard, it’s whether the user understands what they need to do.

anselmolsm, 2016-07-04:
> I think users will pick up what the main toolbar icons do very quickly. If they don’t use the product for a while or it’s their first time, the text is meant to jog their memory. Not to mention [...]

Sorry, but this would receive a [[citation needed]](https://en.wikipedia.org/wiki/Wikipedia:Citation_needed) mark on Wikipedia ;)

> >  Each platform behaves differently. Applications should match the majority of patterns already established by each OS to merge with the environment and decrease the learning curve of users. Both Apple and Google discourage to make a literal port of an application for both systems.

> We just don’t have the resources to implement or test an app which adjusts ui to each OS standard (screen size is hard enough to deal with). Nor do I think it is completely necessary. The question isn’t whether the ui follows a standard, it’s whether the user understands what they need to do.

This discussion is recurrent. I've been seeing this since the times when Symbian was something relevant. At first, the idea of having a single UI to rule them all is compelling, as it gives the impression that it is easier to test and maintain. However, in the long term, the number of adjustments (AKA workarounds) needed to make the UI adapt to different scenarios (e.g. form factors) becomes a pain. If the code architecture is good enough, the initial time to write and maintain UIs for specific cases worth the investment.

Notice I am not saying it is impossible to create a single UI for several different scenarios, I've done this several times. My point is that this approach makes it hard to keep the **quality** in a level higher than PoC / demo. [citation needed] here? Well, been there, suffered that, old times creating [demos](https://qt.gitorious.org/qt-labs/mobile-demos) and applications [targeting](http://www.allaboutsymbian.com/flow/item/16187_Incredible_Circus_comes_to_Bel.php) [different](https://play.google.com/store/apps/details?id=org.indt.circus) [plaforms](https://itunes.apple.com/us/app/incredible-circus/id721643782?mt=8).

DonLakeFlyer, 2016-07-04:
I’m more interested in real users having real problems, understanding those problems and fixing them. And I want to solve those problems in a way that still makes QGC an appealing interface at first glance. There is an element of style to a good user interface as well as usability.

I’ve heard the, you can’t make a single interface across multiple OS thing before. I’ve heard the, you can’t make a single interface that runs at differing screen sizes before. What I haven’t heard is an actual user problem which is important enough to show me that what we have currently is not working. 

As it stands now, we have had one complaint in the last 6 months of someone being confused by the “Q” icon. It’s a good complaint, because the Q icon does not visually indicate anything. I don’t think that warrants having to redo the toolbar ui in a completely different way though. It warrants some tweaking at best. Which is what I have implemented. If that tweaking still doesn’t work, then we can reconsider.

biapalmeiro, 2016-07-04:
@DonLakeFlyer I don't want to be rude, but the tweaking turns the UI kind of amateur and don't solve the problem. This kind of solution won't make an appealing interface at a first glance. If you think our proposal don't solve also, we can work in a better one. We proposed a big change because the toolbar has some problems because of the ammount of information on that, now a days we have buttons, logo, modes, status, notifications. Too many different information displayed together. Changing the header could be a good way to include visual identity and incorporate little by litte what we've seen in the tests.

If you have some metrics about most used mode and platform, for example, we can do a proposal based on that. It can guide if we prioritize iOS or Android elemnets for example. This kind of information is important for us to decide how to change the UI.

About the "Q" icon, it wasn't unsdertood by most of the participants during the test. I don't know what is the icon.

dogmaphobic, 2016-07-04:
:) This goes back to some other discussion I can't remember which. At that time, I was referring to my dislike of compromising the UI on larger screens (tablets/desktop) for the sake of tiny screens (phones).

Elegant, functional, easy. Pick two...

As to explaining the icons:
 * Desktop: Tooltip
 * Mobile: Overlay info screen on first run and have a setting to show it again (as most mobile apps do)

As to the importance of icons, another issued missed above: it helps those who can't read English to have some idea what the thing does.

DonLakeFlyer, 2016-07-05:
"I don't want to be rude, but the tweaking turns the UI kind of amateur and don't solve the problem."

I can buy that, and it’s not rude and likely correct. My main issue is that I don’t think a switch to text is very appealing. Hence I wanted to try some things before jumping all the way back to text only.

Then I get back to my original point which was I wanted to try some things that would work on Mobile/Desktop with the same code. If that is simply not possible, or looks crappy, then I’m fine with Gus’ proposal since it’s a minor deviation in code. And any help overlay thing would still be useful in desktop. Unfortunately last I looked the tooltip support in Qt was really crappy, so we may need to roll our own but not hard. We use to have help overlay for Plan view for example, but we took it out.

Do you think current fading text thing is so bad, we should take it out of 3.0? Anything beyond that is going to have to wait till 3.1.

biapalmeiro, 2016-07-05:
It is bad, sorry... =/ Let us try something simpler than our first proposal. Can we?

AndreasAntener, 2016-07-05:
I'm trying not to argue with you about the "information architecture" approaches discussed here and instead just provide my honest feedback as a user:

I'd like to raise my concern with drop down menus because

> I really don't think people change views too often. When are you planning to release the 3.0? Does telemetry icon exist?

is just not true. If your tweaking the mission or tuning an airframe your constantly switching views. Although this might lead to a completely different discussion about what dev-users need, e.g. a view with everything like we had ages ago, which I'm not going to start now.

I'm very much for an icon approach because I think they are more universally understood. That leads me back to the very beginning and I have to ask why the QGC logo has to be there in the first place (sorry Don ;)? I don't remember any productivity apps (on Mac and IOS) that place their logo in a major menu icon spot.

biapalmeiro, 2016-07-06:
I agree about the logo place. The identity and consistency comes in many different ways: branding color, family icons, all the style that @cfelipesouza is adding little by little.

dogmaphobic, 2016-07-07:
Alright, a bit of an exercise with the main Toolbar taking in consideration we're not going to do any major changes until 3.1.

[image]

For now, the idea is to have some indication of what the *two* main buttons do (Plan and Fly views). For that, I added a (sort of) tooltip. I say "sort of" because tooltips work based on how long you've been *hovering* over an item, which is not possible on touch interfaces. Instead, I will show a tooltip once the item is clicked. It serves the function. It will stay there for a couple of seconds and fade away.

With that said:

* I made the whole toolbar purple, following the guidelines. Sort of. The toolbar is slightly transparent (90% transparent). That caused the original color (#4A2C6D) to brighten as it blended with the background. I reduced the luminance to compensate. I am not yet handling the "light" (outdoor) mode. What color should the toolbar be in that case?
* Setup (the gears) is obvious and I felt it did not need any explanation (tooltip). Same for Preferences (below). It would require the tooltip to be aware of screen limits and position itself accordingly. Way too much work for something that is temporary.
* The logo no longer has any function other than product identification and branding. Yes, logos are used in several apps, especially those that provide similar functionality with similar user interfaces. It is used to quickly identify what app you are using. It could, at some point, bring some *About QGroundControl* dialog, which would be nice to have so the version could be easily checked.
* I moved preferences to its own button on the far right of the toolbar (with its own, new icon). This is what at least some of the apps I use do.

Again, this is all just for 3.0 as we get ready to get it out the door. Eventually, we may need to rethink the UX a bit. Some ideas:

* Maybe vehicle setup should be a category of *Preferences*. In that case, the "dot dot dot" button would turn into *Settings*, which would give you *Application Settings* and *Vehicle Settings*.

* The current *Plan View* should be the *Settings* of a *Mission* type mode (below). In that case, there would be only one view, which is the *Fly View*.

* From the *Fly View* you would have choices (drop down, menu or some other form of selection). The choices would be:

  * User Controlled Flight -- The user is controlling the vehicle (in any of its supported modes such as Manual, Position Control, etc.) and the UI is simply a glorified OSD.
  * Autonomous Flight -- The flight is autonomous, following the settings and choices provided by the user. Here is where you would select *Mission*, *Orbit*, *Follow Me*, etc. As each *Autonomous Mode* is selected, the interface would adapt to that mode (such as showing the current *Plan View* when you select *Mission*).

My thoughts here were to make a clear distinction between when you are flight the vehicle and when the vehicle is doing something on its own. If you think there is a way to make it all one, better.

Does this make sense?

dogmaphobic, 2016-07-07:
> If your tweaking the mission or tuning an airframe your constantly switching views. Although this might lead to a completely different discussion about what dev-users need, e.g. a view with everything like we had ages ago, which I'm not going to start now.

As I suggested above, *Mission* should be a mode of the Fly View (the one and only view).

Tuning an airframe as well as *a view with everything like we had ages ago* are advanced uses. You would basically need two, very different UIs to handle "Normal" and "Advanced". The *view with everything* would never fit a mobile platform either. It's a valid point but short of having two separate UIs, I don't know how you could reconcile both needs.

biapalmeiro, 2016-07-07:
@dogmaphobic It is much better. Just one question, do you think it is possible to apply the header as we initially proposed changing the dropdown to the icons?

biapalmeiro, 2016-07-07:
About your ideas for future releases, I'd prefer discussing them in another issue. There are many items to discuss around that. Anyway, the idea about the views is something that appeared in the tests, so we think it is a perfect way to go.

dogmaphobic, 2016-07-07:
> do you think it is possible to apply the header as we initially proposed changing the dropdown to the icons?

You mean switching away from Icons and using a drop down menu instead?

biapalmeiro, 2016-07-07:
No. Just the spacing, aligments and icons.

biapalmeiro, 2016-07-07:
And do the opposite, switching away the dropdown and using the icons.

dogmaphobic, 2016-07-07:
> No. Just the spacing, aligments and icons.

Uh, yea? I just wasn't planning that much work on this today 8P

I assume you have this all laid out with metrics and icons ready somewhere? Where?

dogmaphobic, 2016-07-07:
At the risk of polluting the thread, here are 3 examples of logos used in the main UI. Just from a few Desktop apps (too lazy to capture screens off my tablet):

[image]
[image]
[image]

biapalmeiro, 2016-07-07:
Yep, not all of them but @cfelipesouza will send you what we have. :)

biapalmeiro, 2016-07-07:
@dogmaphobic The icons for summary are here:
https://github.com/Dronecode/UX-Design/tree/gh-pages/QGC/Application/new_icon_set/summary 

We will share the other there also.

dogmaphobic, 2016-07-07:
Icons for the Summary page will be a bit more involved. Components (and their summaries) are not hard coded. They are loaded dynamically based on the Firmware and/or flight stack. Each component would have to provide QGC with its icon for the summary page. Not complicated but it involves changing a lot of files. I will do that separately (different PR than the toolbar work).

biapalmeiro, 2016-07-07:
Ok. Maybe for 3.1? :)

dogmaphobic, 2016-07-07:
> Ok. Maybe for 3.1? :)

Let me start with the toolbar (once I get the icons). After I get that done, I will start the summary page and see where we get.

dogmaphobic, 2016-07-07:
> I am not yet handling the "light" (outdoor) mode. What color should the toolbar be in that case?

@biapalmeiro How about the color of the toolbar when in Light Mode? Should it be different? I would assume the contrast to be enough. Not to mention when the sun is shining on the tablet, I can't see anything in either mode. That's why I never use the "light" mode. I look for a shade instead 8P

biapalmeiro, 2016-07-07:
The idea is to keep the same toolbar colors. 

The sun shining is a big problem, specially for map view. Don;t know how to solve that yet. :/

AndreasAntener, 2016-07-07:
@dogmaphobic you got me there Gus with the logos although I still think it's unnecessary ;). Anyway, I like the way you did it now.

DonLakeFlyer, 2016-07-07:
* If the Logo really does nothing I'd rather toss it and just go back to the hamburger on the right. I think the hamburger is way more universally known than the dots. We also use hamburger in other places. The dots on the right seem to be an Android only creation. Whereas hamburger is not OS specific and is used on many apps and web sites. Given the usage of the right hand side of the toolbar for communication lost and disconnect messaging I think settings works better on the left.
* Outdoor toolbar needs to stay the same white palette so you can see it. It works well as it is. I've been flying outdoor style for a month now and it works as good as it is going to be. I have ideas to make the map work, but that is another discussion.
* Keep in mind we are just trying to fix a minor usabillity problem with the Q icon not being understood as clickable for Settings. **I don't want much past that for 3.0.**
* I'm specifically not commenting post 3.0 stuff. That's a whole other problem for later.
* For the tooltip kind of things I think it's going to end up being odd given the way they are implemented to have them only on some of the buttons. If they show up after the fact I'd just put big text "Fly", "Plan" overlayed maybe in the top center on the map that fades away. As opposed to connecting it to the button. 

Here is my take for 3.0: Back out the current fading text and just leave everything else alone. I don't think it's a major enough problem to warrant some bigger UI change a few days before I'm trying to cut an RC.  I'm not seeing any solution that is a simple slam dunk at this point. Given the whole flight mode, smart shots stuff and whatever, the whole thing may change anyway. Why take it through a transition that may be short lived.

biapalmeiro, 2016-07-07:
@DonLakeFlyer I understood. But, once @dogmaphobic has done that, why not keep it? Could be exactly as it is, we refine in the future.

App settings and Vehicle setup is kind of a mess. The gear icon is better for app settings (smaller of course and with just 1 gear) and we need another one for vehicle setup. It can solve the 3 dots concern.

What do you think?

DonLakeFlyer, 2016-07-07:
Because I just don't agree with the overall change for which I've detailed mostly above. I don't think it fixes the problem. The two key flaws:
* It doesn't provide a new user with the information of what icons do what until after they click on them.
* I trades a marginal UI concept of a clickable app logo, for another marginal ui concept of dots for settings.
* Plus all the other things I listed above.

closed 2026-10-04 by github-actions[bot]