mavlink/qgroundcontrol #3634: Spektrum bind function doesn't work

kd0aij, 2016-06-24:
Setting parameter RC_DSM_BIND to 0 at the NSH console trigger a bind operation, but using the button on the Radio Calibration panel has no effect.

DonLakeFlyer, 2016-06-24:
What firmware version. I just did this a few days ago without problems.

kd0aij, 2016-06-24:
2.9.7b
Which mode did you select? I note that QGC offers only 2 modes, while the
firmware supports 3.

DonLakeFlyer, 2016-06-24:
I meant firmware version. I'm binding DSMX 8 channels or more. What mode is missing?

DonLakeFlyer, 2016-06-24:
Tested with QGC daily. Firmware bug:
- Flash PX4 master, doesn't work
- Flash ArduPilot Copter 3,3, works

QGC really has little involvement here other than to send a command to the vehicle.

closed 2016-06-24 by DonLakeFlyer

kd0aij, 2016-06-24:
So is this a regression on PX4 master? If so, what was the last version that was tested?
Parameter RC_DSM_BIND can take values 0,1,2
PR https://github.com/PX4/Firmware/pull/4891 would reduce that to a single (universal) choice.

kd0aij, 2016-06-24:
@DonLakeFlyer Is QGC issuing an NSH shell command (like "px4io bind ..." to trigger the bind?
fmu-v4 isn't running the px4io module, so that would explain why it doesn't work on Pixracer.
Which hardware did you test on?

kd0aij, 2016-06-24:
@LorenzMeier Is this function the only one that differs between fmu-v2 and v4? 
Both v2 and v4 need to use the same method to trigger binding, and since there's no px4io app running on v4, that should probably be the one-shot parameter change method.

LorenzMeier, 2016-06-24:
The proper approach would be the command. The parameter change is extremely hacky and should go. Parameters have not been designed to be abused to handle actions :)

kd0aij, 2016-06-24:
OK, I'm not sure where that method came from, but it's also recently been suggested for the WIFI reset function. Should the same command work for v2 and v4 binding? and should it move out of px4io?

DonLakeFlyer, 2016-06-24:
QGC uses MAV_CMD_START_RX_PAIR. I tested on a regular Pixhawk and it didn't work.

kd0aij, 2016-06-24:
https://github.com/PX4/Firmware/issues/4901
Works for me on Pixhawk (fmu-v2)