mavlink/qgroundcontrol #3636: [QGroundControl] Virtual Joystick Radio Roll Calibration not registering

aayush55, 2016-06-24:
I'm using Qgroundcontrol 2.9.7b in linux as my ground station and I want to use the virtual joystick without a real RC or joystick radio by setting COM_RC_IN_MODE to 0(Joystick/No RC Checks). I can arm motors by command line. The problem is when I attempt to takeoff my quapcopter in manual flight mode which runs PX4 flight code, no matter how I move the virtual joystick, the output seems always by 0, I want to know that do I need extra setup before using virtual joystick, or virtual joystick can not control a real quadrotor?

DonLakeFlyer, 2016-06-24:
QGC Daily Build + PX4 Master SITL + virtual joystick is working for me. Trying using the daily build.

aayush55, 2016-06-26:
Thank You DonLakeFlyer,

Its working fine in SITL + virtual joystick,
But I want to control pixhawk using virtual joystick and it is not working.

dagar, 2016-06-26:
@aayush55 Have you set the COM_RC_IN_MODE parameter? SITL configures it for you, but you'll need to set it on a pixhawk.

aayush55, 2016-06-27:
yes I set COM_RC_IN_MODE as 0.
While arming getting error as "ARMING denied manual throttle non-zero" in qgc.
Also in console showing " WARN  navigator timed out "

dagar, 2016-06-27:
I should have been more specific. You need to set COM_RC_IN_MODE to 1 which is what the SITL init scripts do. https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/rcS_jmavsim_iris#L25

anujkumargupta, 2016-09-19:
Hi
I am also trying to connect my pixhawk with qgc using virtual joystick. In manual mode it armed successfully but after giving the takeoff command it says takeoff denied , disarm and retry.I tried again the same thing and i also calibrated all the sensors also but it shows same prob again and again.
Thanks

DonLakeFlyer, 2016-09-19:
> after giving the takeoff command it says takeoff denied , disarm and retry.

The firmware is controlling that. You'll have to ask the firmware folks.

closed 2016-11-15 by DonLakeFlyer

ltrogers98, 2022-02-24:
> > after giving the takeoff command it says takeoff denied , disarm and retry.
> 
> The firmware is controlling that. You'll have to ask the firmware folks

Is there an update on this? Someone I could ask, I'm having the same issue