mavlink/qgroundcontrol #3638: Build MAVLink on top of current binaries, add 2.0 support but default to 1.0

LorenzMeier, 2016-06-24:
@DonLakeFlyer The MAVLink version in QGC is very outdated because it hasn't upgraded in a while. I think the 3.0 release should make the switch to a current version of the library and make the minor adjustments necessary for it to work. That would also allow MAVLink 2.0 receive without signing.

We're already in version hell due to the lack of it as we can't support a number of commands and messages which went into MAVLink in the last weeks. And last but not least it makes QGC not look good if its not proactively participating in the protocol evolution. APM and PX4 both are on the 2.0 version for the release that will ship with QGC 3.0.0.

https://github.com/mavlink/c_library

DonLakeFlyer, 2016-06-24:
I"ll give it a try as long as it's not too involved.

DonLakeFlyer, 2016-06-29:
@LorenzMeier Is there anything to do here after you updated?

DonLakeFlyer, 2016-07-08:
@LorenzMeier Moving to 3.1

DonLakeFlyer, 2016-10-18:
Initial 2.0 support is in

closed 2016-10-18 by DonLakeFlyer