mavlink/qgroundcontrol #3641: Fix unit test framework static delays

NaterGator, 2016-06-24:
The static delays to allow event loops and self-managing objects to init / cleanup is expensive on CI. We should systematically work through and close the loop on all of them so we're not waiting needlessly.

DonLakeFlyer, 2016-06-24:
The one you pointed out is not related to unit testing. It's a general shutdown problem.

NaterGator, 2016-06-25:
I've been looking at this and a huge portion of (in fact, almost all) the overhead comes from synchronizing the start of MockLink and waiting for the parameter load to finish (based on QTimers triggering). I think I'll start there.

DonLakeFlyer, 2016-06-25:
The parameter load under unit test is specifically timed out by MockLink to simulate a real connection. This allows other signals to get out in between parameters coming down from the vehicle. It’s that way on purpose to replicate the normal order of signals when connected to a real vehicle.

DonLakeFlyer, 2016-06-25:
I think it could be sped up, but the actual parameter load unit test itself would need to remain as is with the slower timing. Since it's the only way to test the various failure conditions.

NaterGator, 2016-06-25:
Right, I understand sequencing of signals is critical. One of the things I'm also working on is ensuring deterministic connection of signal/slots before threaded objects are allowed to run their event loop and start generating signals.

As a small example though, every single test has to wait on [this](https://github.com/mavlink/qgroundcontrol/blob/11bcffb0970c34aaff3dc358c975387d29c4f0bc/src/FactSystem/ParameterLoader.cc#L66) timer to expire before the unit test can proceed. The MockLink blasts out its full set of components almost instantly, so this is just wasted 1s per every single individual test run.

DonLakeFlyer, 2016-06-25:
That shouldn't be the case. If all the parameters from the default component are found, the timer is stopped here: https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/ParameterLoader.cc#L226. And the completed signals should go out immdediately in the call to _checkInitialLoadComplete at the bottom.

NaterGator, 2016-06-25:
I know that is the goal, but it isn't working because [this](https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/FactSystem/ParameterLoader.cc#L164) is often seeing [this](https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/FirmwarePlugin/FirmwarePlugin.h#L150) so no default gets set and _checkInitalLoadComplete fails [here](https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/FactSystem/ParameterLoader.cc#L965). 

If we want this to work for the generic firmware case we need to return the PX4 default parameter ID name from the FirmwarePlugin base class since PX4 parameters are [used](https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/comm/MockLink.cc#L187) in the generic case.

DonLakeFlyer, 2016-06-25:
Ah, there's bug in the logic here: https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/FactSystem/ParameterLoader.cc#L223. There should be an "or no default component id param known" check there as well.

DonLakeFlyer, 2016-06-25:
```
  } else if (_defaultComponentId != MAV_COMP_ID_ALL || _defaultComponentIdParam.isEmpty()) {
 ```

NaterGator, 2016-06-25:
That is even worse... then the tests will never complete because the timer is stopped and the condition [here](https://github.com/mavlink/qgroundcontrol/blob/714b2e168eda6e3af8d312e01b37f64cd8acab06/src/FactSystem/ParameterLoader.cc#L965) will stop the final param of a generic test from emitting that they are loaded.

DonLakeFlyer, 2016-06-25:
The generic mavlink, no default component id param known, case is pretty screwed up then. I'm not sure if the similar or check is needed there as well. Would have to stare at it some more. The code has to fight with the fact that the mavlink spec has a hole in it with respect to determining the default component id. The default component id param is just a hack that works for known firmwares. But for unknown firmwares, there really is no way to know correctly.

NaterGator, 2016-06-26:
I believe [this](https://github.com/mavlink/qgroundcontrol/pull/3653/commits/14c9c86e16ef1885b23c84059695c72d252dd97c) cleanly fixes the default component generic case.

DonLakeFlyer, 2016-06-26:
Yeah, I think so Nate. Awesome. Let me let through it all again one more time and I"ll merge.

NaterGator, 2016-06-26:
Wait a bit on merging if you don't mind. I want to pour over the tests a bit more and discuss the implications of [this](https://github.com/mavlink/qgroundcontrol/pull/3653/commits/931aad56fad37be4f88c40b293955247d3e7ad03#diff-b470bec35d98cca1c1bd05154d315fa8R273) in a bit more detail. First I need to eat dinner :wink:

DonLakeFlyer, 2016-06-28:
Was there some other explanation you wanted to add here?

NaterGator, 2016-06-28:
Sorry, yes. On the subject of deleteLater() for links... 

Obviously you're familiar with the tired "you're doing it wrong" blog post with respect to the Qt threading API (and, subsequently their revisions to the documentation that in fact you're not _necessary_ doing it wrong). 

It really is best practice with QObjects to deleteLater instead of delete directly; with the links we are often calling moveToThread(this) before starting the threads own event loop. A bit illogical, sure, but not semantically or effectively wrong. We just need to be careful that any threaded link makes a pair of calls like [here](https://github.com/NaterGator/qgroundcontrol/blob/931aad56fad37be4f88c40b293955247d3e7ad03/src/comm/MockLink.cc#L80) and [here](https://github.com/NaterGator/qgroundcontrol/blob/931aad56fad37be4f88c40b293955247d3e7ad03/src/comm/MockLink.cc#L133) to ensure the threaded LinkInterface object is always actually managed by a running event loop.

closed 2024-06-02 by HTRamsey