mavlink/qgroundcontrol #3651: Windows Store Build

dagar, 2016-06-25:
We have QGroundControl reserved for the Windows store, now we just need an appropriate build to submit.

I think having a regular desktop build that doesn't update, and a presence in the store is a good way to split users that want auto updates, and those that don't (professionals, etc).

Windows is by far the most popular platform for QGroundControl. Looking over the download stats it's > 60%.

DonLakeFlyer, 2016-06-25:
How did you get the download stats out of S3? I was trying to do that and couldn't find it.

Also the 3.0 release should be the first release to the store. No reason to wait till 3.1

dagar, 2016-06-25:
I was just going by Github releases. http://www.somsubhra.com/github-release-stats/?username=mavlink&repository=qgroundcontrol
We'd have to add our own tracking to get s3 stats.

dagar, 2016-06-25:
Happy to do it for v3.0 if we can get working.

dagar, 2016-07-08:
We'll need that Windows team developer account to proceed.

DonLakeFlyer, 2016-07-10:
Lorenz is getting a company account set up.

jcooper21, 2016-08-16:
Any news on this?

DonLakeFlyer, 2016-08-18:
Still no progress.

DonLakeFlyer, 2017-07-07:
Going to have to wait again. Don't have time to deal with it now.