mavlink/qgroundcontrol #3662: RadioConfigTest Failure OSX

dagar, 2016-06-26:
RadioConfigTest is consistently failing on OSX in the builds. Any ideas?

https://travis-ci.org/mavlink/qgroundcontrol/jobs/140290756

********* Start testing of RadioConfigTest *********
Config: Using QtTest library 5.5.1, Qt 5.5.1 (x86_64-little_endian-lp64 shared (dynamic) release build; by Clang 6.0 (clang-600.0.56) (Apple))
PASS   : RadioConfigTest::initTestCase()
QDEBUG : RadioConfigTest::_fullCalibration_px4_test() Temp log "/var/folders/my/m6ynh3bn6tq06h7xr3js0z7r0000gn/T/FlightData452750.mavlink419981" false
QDEBUG : RadioConfigTest::_fullCalibration_px4_test() QGCMessageBox (unit testing) "Unit Test" "Error during Mission communication with Vehicle: Vehicle did not respond to mission item communication: MISSION_COUNT"
FAIL!  : RadioConfigTest::_fullCalibration_px4_test() Compared values are not the same
   Actual   (_missedMessageBoxCount): 1
   Expected (0)                     : 0
   Loc: [/Users/travis/build/mavlink/qgroundcontrol/src/qgcunittest/UnitTest.cc(137)]
QDEBUG : RadioConfigTest::_fullCalibration_apm_test() Temp log "/var/folders/my/m6ynh3bn6tq06h7xr3js0z7r0000gn/T/FlightData452750.mavlink404608" false
PASS   : RadioConfigTest::_fullCalibration_apm_test()
PASS   : RadioConfigTest::cleanupTestCase()
Totals: 3 passed, 1 failed, 0 skipped, 0 blacklisted
********* Finished testing of RadioConfigTest *********

DonLakeFlyer, 2016-06-26:
I'll take a look. Likely caused by recent parameter loader unit testing changes.

DonLakeFlyer, 2016-06-26:
Of course only failing on travis. Fine on my machine.

DonLakeFlyer, 2016-06-26:
I think I figured this out. It's a timing bug introduced with my pull to load items automatically into plan view.

DonLakeFlyer, 2016-06-26:
Nope, still haven't figure this out.

DonLakeFlyer, 2016-06-26:
If you re-run you have a 50/50 chance of getting it to pass.

NaterGator, 2016-06-26:
I'm getting this too on Windows 10.

dagar, 2016-06-26:
Seems to be resolved as of https://github.com/mavlink/qgroundcontrol/commit/59ef1b3bd634762fa7d80c7f8b3143d95b1973b0

closed 2016-06-26 by dagar

DonLakeFlyer, 2016-06-26:
Yeah. Another signaling problem with unit tests. I'm going to have to clean up all the signaling between MultiVehicleManager, Vehicle, ParameterLoader, MissionManager. It's turned into quite the nightmare.