mavlink/qgroundcontrol #3671: No Channel Option for Channel 6

Rudi3, 2016-06-26:
In the current stable release I cant choose a channel option(In flight modes) for channel 6. It could be intentional for some reason, im not sure, but I'm missing it as I'm losing a flight mode switch. (But its there for gimbal control). And it really looks like there is something missing: 5, ?, 7, 8, 9,...

[image]

DonLakeFlyer, 2016-06-27:
There is no such thing as a channel 6 option. It's reserved for gimbal I think.

closed 2016-06-27 by DonLakeFlyer

Rudi3, 2016-06-27:
http://ardupilot.org/copter/docs/ac_rollpitchtuning.html Actually there is a channel 6 option and why should you reserve a channel for gimbal? You can freely pick the gimbal channels.

DonLakeFlyer, 2016-06-27:
QGC doesn't support channel 6 tuning yet. We'll add it as some point.

Rudi3, 2016-06-27:
Oh ok nice :smile:

DonLakeFlyer, 2016-06-28:
#3675