mavlink/qgroundcontrol #3674: Error connecting: Could not create port.

Inakimalerba, 2016-06-27:
I'm getting this error when trying to run the latest version of qgc (2.9.7b-365-gc87b34b) on Debian 8 and Ubuntu 14.04.

> Error on link /dev/ttyUSB0. Error connecting: Could not create port. Permission error while locking the device.

I already did add the user to dialout and remove modemmanager.

BTW. I also have the version 2.9.2 and it works without

nerocide, 2017-02-02:
I spent couple of hours on the exact same problem.
I was actually trying to **connect** my 3DR radio devices (chinese cheap version) **manually while the autoconnect was enable for SiK radio in QGC**.

When the feature was activated and ground module plugged in, the lock would be present, like so:
`user@kali:/run/lock$ ll
total 4
-rw-r--r-- 1 user user 25 févr.  2 23:14 LCK..ttyUSB0
`
Meaning  that a manual re-lock was impossible and would raise the error....which makes sens.

This article helpt me also finding and setting a correct baudrate for both air and ground module.
[http://ardupilot.org/copter/docs/common-3dr-radio-advanced-configuration-and-technical-information.html](url)
Hope it will help you.

DonLakeFlyer, 2017-02-21:
Closing. Covered in better named issue

closed 2017-02-21 by DonLakeFlyer

Aniket0022, 2018-02-14:
I had issue with serial port permission,
Solution :
      Find the process id of that app, by command :
      Ps ax
Then from that list search your applixation, in my case it was serialPortTs with pid 1978
Then write as :
  Kill - 9 1978
Thats it.. You have to kill and reboot the system

imrasp, 2018-07-09:
Sadly, I have this problem on 2018 but I finally fix this by disable auto connect in QGC (application setting/general). `LCK..ttyUSB0` can help me fine which process is lock the port If I understand it correctly.