mavlink/qgroundcontrol #3675: Support channel 6 tuning

DonLakeFlyer, 2016-06-27:
http://ardupilot.org/copter/docs/ac_rollpitchtuning.html

Rudi3, 2016-06-29:
@DonLakeFlyer It seems that I made it work! may I make a pull request? (I kind of have no rights)

DonLakeFlyer, 2016-06-29:
You don't need any rights to submit a pull request. Anyone can do it. Is it not working for you?

Rudi3, 2016-06-29:
@DonLakeFlyer Sorry I'm still new to git/github. I'll read more docs and try again. Thanks!

DonLakeFlyer, 2016-06-29:
I can walk you through it. It's fairly easy. Won't have time till later though to write it up

DonLakeFlyer, 2016-06-30:
1) Fork your own copy of the repository on GitHub
2) git clone that to your local machine
3) Add an upstream remote "git remote add upstream https://github.com/mavlink/qgroundcontrol.git"
4) Make you changes on your clone
5) Run the unit tests
6) Push you changes to your fork
7) Go to https://github.com/mavlink/qgroundcontrol.git in GitHub

You should now see an option to create a pull request

Rudi3, 2016-06-30:
@DonLakeFlyer Thanks again, I forgot to fork :fork_and_knife: :sweat_smile: 
Somehow I wasn't able to run unit tests on my machine... the --unittest option seemed to be ignored (on the vagrant build)... But Travis should do that now

closed 2016-08-19 by DonLakeFlyer