mavlink/qgroundcontrol #3676: QGC - OS X crash on startup

lvale, 2016-06-28:
Latest download of QGC for OS X 10.11.5

When starting QGC always crashes and we get the system message that QGC crashed and Ignore/Report/Reopen options. Crash was always on Thread0

After some troubleshooting with the help of Don, I managed to isolate the following line on .config/QGroundControl.ini that causes the crash

MaxDiskCache=4096

After deleting this line, QGC started without issues.

The .config files are attached
[org.qgroundcontrol.zip](https://github.com/mavlink/qgroundcontrol/files/336259/org.qgroundcontrol.zip)

dogmaphobic, 2016-06-28:
The default max disk cache is set to 1,024 MB. Did you set it to 4,096 (4GB) yourself? I have mine set to 16G and it runs fine. I have to assume you have more the 4G free disk space?

lvale, 2016-06-28:
Yes, manually changed to 4096, and yes have more than 4G space left (about 500G)

dogmaphobic, 2016-06-28:
Hmm, I can't repro this. I set the limit to exactly 4096 as you had and it all still works. I removed the original cache to start over from scratch and it all still works as expected. When you get the crash, do you see a window that looks like the one below? Could you please cut and paste its contents so I can have some idea where the crash is happening?

[image]

lvale, 2016-06-28:
Sure.

btw: Now I can replicate the issue easily just by replacing the .config contents with the one attached above

[image]

Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.3 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.5f3 17904, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en3
Network Service: Wi-Fi, AirPort, en0
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM1024F, 1 TB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5

NaterGator, 2016-06-28:
Forgive me for being naive and jumping in here, but from a distance 4GB looks an awful lot like the 32 bit memory address space for OS X. Does the sqlite driver use memory mapped IO (or are we loading cache into memory)?

dogmaphobic, 2016-06-28:
I meant the entire content of that screen (Select all, copy and paste it here). Not the screen grab. I used a screen grab so it could easily be identified.

> 4GB looks an awful lot like the 32 bit memory address space

That was my first thought. I've been running with it set to 16G with no issues. SQLite uses 64-bit integers but there could be something in between using uint32_t for some reason.

> The maximum size of a database file is 2147483646 pages. At the maximum page size of 65536 bytes, this translates into a maximum database size of approximately 1.4e+14 bytes (140 terabytes, or 128 tebibytes, or 140,000 gigabytes or 128,000 gibibyte).

lvale, 2016-06-28:
Attached the crash report (sorry) and the contents of .config/org.qgroundcontrol

[org.qgroundcontrol.zip](https://github.com/mavlink/qgroundcontrol/files/337782/org.qgroundcontrol.zip)

[crash log.txt](https://github.com/mavlink/qgroundcontrol/files/337780/crash.log.txt)

dogmaphobic, 2016-06-28:
Found it. It's not the disk cache, it's the memory cache. You had it set to 2G and Qt is using signed 32-bit integers somewhere when handling memory cache. I will force it to be less than 2G.

I handle disk cache but memory cache is handled internally by Qt.

```
MaxDiskCache=4096
MaxMemoryCache=2048    <== This is the cause
```

NaterGator, 2016-06-28:
Actually that's on us. We're performing an implicit type conversion from [here](https://github.com/mavlink/qgroundcontrol/blob/f0b7195e9e10267bef440a425274e8b93c40a6fe/src/QtLocationPlugin/QGeoTiledMappingManagerEngineQGC.cpp#L218) to [here](http://code.qt.io/cgit/qt/qtlocation.git/tree/src/location/maps/qgeotilecache_p.h?h=v5.5.1#n121).

dogmaphobic, 2016-06-28:
Oh, why, why would you ever use a signed integer for something that should be size_t? (rhetorical question. It's Qt after all) 8P

dogmaphobic, 2016-06-28:
Hopefully fixed in #3684

closed 2016-06-28 by dogmaphobic

DonLakeFlyer, 2016-06-29:
> Oh, why, why would you ever use a signed integer for something that should be size_t? (rhetorical question. It's Qt after all) 8P

That has always made me a bit nuts as well.