mavlink/qgroundcontrol #3677: ArduPilot Parameters Descriptions and Values not being used

lvale, 2016-06-28:
When editing values on full parameter list the right dialog is not being populated with the Description and Values and Reboot required 

example

[image]
  

  // @Param: PWM_COUNT
    // @DisplayName: Auxiliary pin config
    // @Description: Control assigning of FMU pins to PWM output, timer capture and GPIO. All unassigned pins can be used for GPIO
    // @Values: 0:No PWMs,2:Two PWMs,4:Four PWMs,6:Six PWMs,7:Three PWMs and One Capture
    // @RebootRequired: True

DonLakeFlyer, 2016-06-28:
Need to know what firmware type Copter/Plane and what version 3.3/3.4/etc?

DonLakeFlyer, 2016-06-28:
Also ArduPilot does not have default value support.

lvale, 2016-06-28:
@DonLakeFlyer I believe this is across all variants, The example above is taken from Copter 3.4RC1 but it's not relevant, because other params have similar behavior.

closed 2016-06-29 by DonLakeFlyer

DonLakeFlyer, 2016-06-29:
Amazing that the long description has been busted forever!

lvale, 2016-06-29:
Mea culpa, mea culpa....It's a matter of looking and not seeing ;)