mavlink/qgroundcontrol #3682: The distance about planned home position is minus

glody, 2016-06-28:
Test Environment
QGroundControl: Daily Build: v2.9.7b-335-g4f119b3
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the tips about planned home position is correct
Step:
1. Lanuch QGC
2. Check the distance tips on mission page
Excepted result:
The distance about home position should be 0M
Actual result:
The distance is -1M, it doesn't make sense

[image]

DonLakeFlyer, 2016-06-29:
@glody Can you make sure you are doing your testing on the latest daily build instead of Stable. Stable is very old now. So this is different now. It actually exposes a different bug!

[image]

closed 2016-06-29 by DonLakeFlyer

glody, 2016-07-04:
@DonLakeFlyer Yes, I am using daily build for our testing. I found this issue on daily build 24/06/2016.

Re-check on daily build 01/07/2016, this issue still able to reproduce. the build version is v2.9.7b-379-g2e9d112.

Reprodue steps:
1. Set some waypoints
2. Select home point
3. Check "distance" shows in tips

[image]

DonLakeFlyer, 2016-07-07:
This is fixed now. For real!