mavlink/qgroundcontrol #3683: The estimated map size about offline map has large difference with real size

glody, 2016-06-28:
Test Environment
QGroundControl: Daily Build: v2.9.7b-335-g4f119b3
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the estimated map size is accurate.
Step:
1. Lanuch QGC
2. Click "Add new set" in Offline Maps item Preference page
3. Zoom the the max and min map size and check the "Set Size(Est.)"
4. Click Download
5. Check the real set size.
Excepted result:
The estimated set size should almost same with real size.
Actual result:
Here has large difference between estimated size and real size.

[image]

[image]

dogmaphobic, 2016-06-29:
There is no way to tell what the final size will be. Each tile will have different sizes based on how well they compress. If you get large swaths of ocean, the final size will be a lot smaller as those (mostly all blue) tiles will smaller than the average. I use an pre-computed, worst case average and assume a worst case scenario. That way, you don't run out of disk space in case you get a region with large changes (and little compression).

DonLakeFlyer, 2016-06-29:
Sound like it's as good as it's going to get

closed 2016-06-29 by DonLakeFlyer

dogmaphobic, 2016-06-29:
There used to be some verbiage explaining that the estimated value was a worst case scenario and that the final size would only be known once it was all downloaded. It was fine for the desktop version but we certainly don't have the space for it on mobile devices.