mavlink/qgroundcontrol #3696: Android instrument panel freezing

jcooper21, 2016-06-29:
On android 6 on a nexus 6 the instrument panel freezes after about 10 minutes of flight and app had to be restarted and the 915 MHz telemetry disconnected and reconnected.

DonLakeFlyer, 2016-06-30:
Were you running a mission?

jcooper21, 2016-06-30:
I was flying in altitude control mode and stabalize but used auto take off.

DonLakeFlyer, 2016-06-30:
Ok, thanks.

DonLakeFlyer, 2016-06-30:
Likely related to #3698

DonLakeFlyer, 2016-06-30:
@dogmaphobic Do you know how an ANR is reported by the user. Do they get prompted to send info up to developer. There are no ANRs reported at all against QGC Android builds.

dogmaphobic, 2016-06-30:
I had to Google it to find out what an ANR was, let alone knowing what it does :)

Based on this:

https://developer.android.com/training/articles/perf-anr.html

It seems it simply asks the user if they want to shut down the app.

DonLakeFlyer, 2016-07-08:
@jcooper21 Did you get a dialog that asked you about reporting the hang? If you do make sure the send it up. Then I can get a stack trace from your device.

DonLakeFlyer, 2016-07-08:
Can you also try latest Daily Build. Map flight trails were chewing memory. That should be fixed now. If we're lucky this may fix your problem.

jcooper21, 2016-07-08:
I won't be able to test for a couple weeks I'm on vacation in Mexico.

DonLakeFlyer, 2016-07-08:
I ran Android against SITL on a looping mission for over an hour without seeing a hang or any memory leaking.

DonLakeFlyer, 2016-11-10:
@jcooper21 Are you still seeing this?

jcooper21, 2016-11-10:
I have not seen it lately.

closed 2016-11-10 by DonLakeFlyer