mavlink/qgroundcontrol #3698: Memory during flight

DonLakeFlyer, 2016-06-30:
- Setup a looping mission
- Crank it up in SITL against an iOS build
- Wait

Watch the memory usage, go up and up. Doesn't happen really fast, but still a bad leak.

DonLakeFlyer, 2016-06-30:
The memory increase is from the trajectory lines. Without trajectory lines memory is very stable.

NaterGator, 2016-06-30:
I was looking at this today. Is there a particularly good reason to store these as coordinate vectors? Simply storing a QVariantList of QGeoCoordinates can be bound directly to a MapPolyLine without any of the model/view/qobject overhead. 

Also, it may not be a bad idea to have a setting for the maximum "age" of the trajectory line. 5 minutes sampled at 1 second could be reserved up front and would only require 5*60*sizeof(QGeoCoordinate) bytes.

DonLakeFlyer, 2016-06-30:
Yeah, I don't know. This trajectory line stuff is pretty old. So it may be out of what with current understanding of what you can do with MapPolyLine. I was planning at looking at Mission Planner to get a sense for how it handles maximum age of lines.

DonLakeFlyer, 2016-06-30:
> Simply storing a QVariantList of QGeoCoordinates can be bound directly to a MapPolyLine without any of the model/view/qobject overhead.

That means the full list needs to be signalled for change each time a point is added though. Whereas using a ListModel you get incremental update. Correct?

DonLakeFlyer, 2016-06-30:
Also MapPolyLine doesn't support a model, it only supports a list. So you can't have a single ListModel of QGeoCoordiantes that updates passed to MapPolyLine.

DonLakeFlyer, 2016-06-30:
Which I think means there isn't really a better way to do it?

DonLakeFlyer, 2016-06-30:
@NaterGator Please let me know if you still think there is a different way to do this. Had to look and write things down to re-create in my head why I did it this way.

DonLakeFlyer, 2016-07-01:
MP has a track length settings that defaults to 200. I assume that means 200 segments.

NaterGator, 2016-07-01:
I'm on vacation which means I've had a craft beer or two, but let me dig into the Qt side code and see how we can go about this in the least expensive way. My gut is that we should have a time limit (maximum upward bound of number of line segments) and then allocate the correct number of CoordinateVectors. Store the pointers in a QQueue and update the coordinates directly without managing memory. That should save triggering model updates and directly update the individual MapPolyLine without causing much memory fragmentation. But again, I'll dig around the Qt source a bit and see if I think we can do better.

DonLakeFlyer, 2016-07-01:
@NaterGator I'm going to live with the current implementation and the new time limit for 3.0. Could you enter an Issue for 3.1 to revisit this implementation?

Happy vacation! Drink one for me.

NaterGator, 2016-07-01:
Sure. In the mean time I'll also submit a PR with the aforementioned implementation for people to test if they feel so inclined.

DonLakeFlyer, 2016-07-01:
Seems to leak if you run a radio as opposed to WiFi. Still testing.

DonLakeFlyer, 2016-07-01:
Looks like this is still the trajectory points causing the problem even though they are limited now.

DonLakeFlyer, 2016-07-01:
Still can't repro with SITL.

DonLakeFlyer, 2016-07-01:
Something internal to Qml is leaking. I can see it using OSX Instruments and take generation dumps. Don't think it likes the way we are updating things. Possibly memory marked wrong.

DonLakeFlyer, 2016-07-01:
I'm going to try moving trajectory handling all inside the Qml. Then I can use MapPolyLine.[add|remove]Coordinate.

DonLakeFlyer, 2016-07-06:
And of course MapPolyLine.addCoordinate is buggy as hell as well!

NaterGator, 2016-07-07:
@DonLakeFlyer I am surprised it is that bad. The [implementation](http://code.qt.io/cgit/qt/qtlocation.git/tree/src/imports/location/qdeclarativepolylinemapitem.cpp?h=v5.5.1#n561) is fairly straightforward (if not likely nearly as expensive).

When you said something internal to Qml is leaking can you elaborate on what you modified? Is that pushed somewhere?

DonLakeFlyer, 2016-07-07:
It always draw a line going off to to never never land every once in a while even though the coordinates are correct. I've tried doing it multiple ways and still the same problem of lines goind crazy directions.

The currect code leak is somewhere inside Qt which I can't quite track down. It's memory passing over to the Qml side which isn't getting freed. I'm going to try the current code concept using a fixed allocation to see if it fixes it.

NaterGator, 2016-07-07:
I see. For what it is worth setting an entire list is [about](http://code.qt.io/cgit/qt/qtlocation.git/tree/src/imports/location/qdeclarativepolylinemapitem.cpp?h=v5.5.1#n541) as expensive as add/remove. I'm going to take a swipe at this now if that's ok. I don't have any hardware at home though, so I may need you to test to see if things are still leaking.

DonLakeFlyer, 2016-07-07:
I tried setting the entire list in a single MapPolyLine and singalling full list change on each update. It doesn't work. It draws crazy lines off into nowhere. You can test with SITL and a mission or just GoTo location to get some lines to draw. You can't memory test with SITL through, which I don't quite understand.

DonLakeFlyer, 2016-07-07:
After days of looking at this I'm pretty sure this is just chasing Garbage Collector ghosts. No problem here now that it limits the segments. But we did discover all the way MapPolyLine falls on it's face!

closed 2016-07-07 by DonLakeFlyer