mavlink/qgroundcontrol #3710: Add support for Motor Test

DonLakeFlyer, 2016-07-01:
ArduPilot can already do this. Not sure about PX4.

dagar, 2016-07-01:
I'm guessing MAV_CMD_DO_MOTOR_TEST in mavlink ardupilotmega? I can do the PX4 side if we can agree on how it should be done. Can we get away with moving a mavlink cmd from ardupilotmega -> common?

LorenzMeier, 2016-07-02:
@dagar Moving commands should be possible, yes.

dagar, 2016-07-03:
Moving MAV_CMD_DO_MOTOR_TEST to common: https://github.com/mavlink/qgroundcontrol/issues/3710

DonLakeFlyer, 2016-07-10:
FYI: Working on this now.

[image]

DonLakeFlyer, 2016-07-10:
Is there a mavlink message or something which comes back from the Vehicle which tells me what the motors are actually doing?

dagar, 2016-07-10:
SERVO_OUTPUT_RAW?

DonLakeFlyer, 2016-07-10:
Do you want me to add this to the PX4 wide as well in QGC? It will just generate a command not supported until the firmware catches up.

dagar, 2016-07-10:
It's not in common yet. Can you review/merge? https://github.com/mavlink/mavlink/pull/580

dagar, 2016-07-10:
Yes, please add it for PX4 as well. Then I won't have to hack something to send the command.

https://github.com/PX4/Firmware/issues/5023

DonLakeFlyer, 2016-07-10:
Something that would be useful is a way to determine motor count from mavlink somehow. I can hardwire somewhat based on MAV_TYPE but then that gets into VTOL and so forth and all hell breaks loose.

dagar, 2016-07-10:
PX4 airframe metadata?

jaxxzer, 2017-03-08:
I would suggest having as many sliders as the board has motor outputs, regardless of frame. Invalid motors should be handled on the autopilot and do nothing if they are not actually configured for motor output according to the vehicle type.

DonLakeFlyer, 2017-03-08:
> PX4 airframe metadata?

Not very helpful since firmware specific.

KJBB, 2018-10-08:
Keen on an individual motor test, I get stymied at each turn. The link https://github.com/bluerobotics/qgroundcontrol/tree/motor-test gives a 404 Not Found. https://www.dronecode.org/category/releases/qgroundcontrol-release/ gives V3,2 as the highest release number, but I am running V34.1, and still don't see a motor test. The last comment above, was about 19 months ago. Can you clarify the status of motor testing as of - Oct 2018?

AK7466, 2019-01-21:
Like KJBB (and presumably many others) I too am still stymied!!

Has this really not been resolved yet? Now mid January 2019

Thanks A.K.

DonLakeFlyer, 2019-01-21:
Still waiting on firmware support...

AK7466, 2019-01-21:
' Guess we'll just have to be patient.........

jaxxzer, 2019-01-22:
Relevant [mavlink](https://github.com/ArduPilot/mavlink/pull/27) and [ardupilot](https://github.com/ArduPilot/ardupilot/pull/8202) prs (only ardusub supports this, the rest of the vehicles use [this](http://ardupilot.org/copter/docs/connect-escs-and-motors.html#checking-the-motor-numbering-with-the-mission-planner-motor-test) method). I don't know where PX4 is with this.

Here are the qgc prs on this in the past:
#3786
#3858
#6912

DonLakeFlyer, 2019-03-22:
Still waiting on firmware support

AK7466, 2019-03-23:
Me to!…………….

MatteoLanni, 2020-06-24:
Any update? June 2020 and I can't use motor test on PX4 1.10.1

DonLakeFlyer, 2020-06-24:
This has been supported for PX for a while now. You need to be using a version of the firmware which supports it. Not sure which version it started with. You'll have to ask the firmware folks.

closed 2020-06-24 by DonLakeFlyer

AK7466, 2020-06-24:
Thanks.

A.K.
---------------------------------------------------------------------------------------------------