mavlink/qgroundcontrol #3719: Joystick Setup Issues

dogmaphobic, 2016-07-02:
Joystick setup layout needs some work:
You can no longer flick the screen, therefore I'm not able to reach all the buttons.
Button monitor boxes end at 15 and does not show all buttons. Layout is a bit all bunched up too making boxes with more than one digit too small for the text in holds.

[image]

DonLakeFlyer, 2016-07-02:
Of course Gus has a joystick with 72 buttons on it!

dogmaphobic, 2016-07-02:
It's a flight simulator "Joystick". As is, it provides 38 buttons and 6 axis. I didn't plug in the rudder pedals, which would make it even worse 8P

DonLakeFlyer, 2016-11-24:
I fixed this a while back. All setup pages are now in a QGCFlickable and the button monitor is a Flow.

closed 2016-11-24 by DonLakeFlyer