mavlink/qgroundcontrol #3722: PX4 Spektrum DX8 radio setup: roll and yaw channels reversed

dagar, 2016-07-03:
Pitch + throttle are correct, but maybe they didn't need to be reversed.
QGC + PX4 master

DonLakeFlyer, 2016-07-07:
Can you attach the param set? What type of vehicle?

dagar, 2016-07-07:
It was the HIL easystar. I'll reset all params and do a fresh setup tomorrow.

dagar, 2016-07-08:
Something is very wrong with aileron and yaw. I reset all params, used latest QGC and flashed PX4 master.

https://gist.github.com/dagar/36d311a3a77f8fe3ebfc43f5ace6a418

dagar, 2016-07-08:
All the sticks are neutral, but aileron and yaw are wrong.

[image]

DonLakeFlyer, 2016-07-09:
Ok. Much strangeness going on here. Let me pull out my DX8 and try to repro.

DonLakeFlyer, 2016-07-09:
I broken the channel monitor displays when I updated the pwm range to be wider. If the channel was reversed they would not display correctly. The Yaw and Roll channel are correctly being marked as reversed. Can you try again to make sure it's all working now.

dagar, 2016-07-11:
The display is sane again, but roll/yaw are still not being reversed. PX4/QGC latest master + completely wiped pixhawk and Spektrum DX8.

DonLakeFlyer, 2016-07-11:
In the param set you attached above Roll and Yaw parameter show reversed.

dagar, 2016-07-11:
hmm, I didn't even test fly that one because the display was so wrong.
Here's the one I just did. https://gist.github.com/dagar/23171fcbb0db4c429048b949ec76eeaf
Roll and yaw are reversed flying in HIL. Navigation still works so it's definitely not a problem with the HIL model, xplane, etc.

DonLakeFlyer, 2016-07-11:
Params aren't reversed. The pull I put through was ui only change. Strange.

DonLakeFlyer, 2016-07-11:
I just tried it again using PX4 Beta and Stable_V3.0 build and the correct channels were reversed. Can you do the same with RadioComponnentControllerLog on and post the results.

dagar, 2016-07-12:
Same result.
QGC: Stable_V3.0 PX4: beta
RadioComponentControllerLog output
https://gist.github.com/dagar/314e95c7f0b152722c490c591dfdf9d1

Do I have something misconfigured? I reset my DX8 to defaults.

DonLakeFlyer, 2016-07-12:
According to the log Roll and Yaw are being correctly seen as reversed. Yet the params for them are not marked as reversed (which I can't repro with my DX8).

The log output to be looking for is here:
```
RadioComponentControllerLog: _inputStickDetect function:channel:value RC_MAP_ROLL 0 1100
RadioComponentControllerLog: _inputStickDetect function:channel:value RC_MAP_ROLL 0 1100
RadioComponentControllerLog: _inputStickDetect settle complete, function:channel:value:reversed 0 0 1100 true
```

Can you double-check the params again after the cal to make sure they are incorrect (yeah I know, yet again). If so, I'm going to need to add some more logging to track this down.

dagar, 2016-07-12:
Weird. Is completely wiping the params each time actually of any value?

DonLakeFlyer, 2016-07-12:
Not needed.

DonLakeFlyer, 2016-07-23:
@dagar Any more new on this?

dagar, 2016-07-24:
Sorry, lost track of this. Trying again.

dagar, 2016-07-24:
Still the same with QGC Stable_V3.0 and PX4 Beta.

dagar, 2016-07-24:
I think it's this.
https://github.com/mavlink/qgroundcontrol/blob/73bf8300681de79e7d409a9f515cd3bd98ebb083/src/AutoPilotPlugins/Common/RadioComponentController.cc#L792

dagar, 2016-07-24:
Am I missing something in my understanding of RC input, mixer, pwm output, or is that multirotor check bogus?

DonLakeFlyer, 2016-07-25:
Two problems:
1) The if structure is wrong in that it nevers writes a reversed default when non-MR. Then there is the question of determining reversed on non-MR vehicles. 
2) I think there may be a PX4 versus ArduPilot problem there but not sure yet. For PX4 if you need to reverse channel output due to servo direction you have to do that in the mixer right. You DO NOT do it by reversing the transmitter.

DonLakeFlyer, 2016-12-08:
@dagar Fixed wing ends up being quite confusing given servo control horn setup and direction. The thinking was that for FW you need to adjust the parameter values and radio reverse settings manually to get correct movement. You can't do it automatically, you actually need to see the surfaces move noth in Manual and Stablized modes.

dagar, 2016-12-09:
Right, but that's completely separate from input (at least in PX4). With the current setup you can end up with both reversed and think you're fine in manual testing, and some stabilized if you don't also move the plane.

DonLakeFlyer, 2016-12-30:
Ok, so for PX4 you think Radio Cal should set reversed param for all vehicle types. Correct?

dagar, 2016-12-30:
Correct

DonLakeFlyer, 2016-12-30:
@dagar Can you retry when this goes through?

closed 2016-12-30 by DonLakeFlyer