mavlink/qgroundcontrol #3724: Error: too many arguments to function 'uint16_t mavlink_finalize_message_chan

TSC21, 2016-07-04:
Getting this after updating to latest master, compiling using QT Creator 5.5.1. Submodules already updated with `git submodule update`, but I guess this is `mavlink` module related.

```bash
compiling ../qgroundcontrol/src/comm/MAVLinkProtocol.cc
../qgroundcontrol/src/comm/MAVLinkProtocol.cc: In member function 'void MAVLinkProtocol::_sendMessage(LinkInterface*, mavlink_message_t)':
../qgroundcontrol/src/comm/MAVLinkProtocol.cc:449:169: error: too many arguments to function 'uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)'
     mavlink_finalize_message_chan(&message, this->getSystemId(), this->getComponentId(), link->getMavlinkChannel(), message.len, message.len, messageKeys[message.msgid]);
                                                                                                                                                                         ^
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/../protocol.h:85:0,
                 from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h:30,
                 from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/mavlink.h:25,
                 from ../qgroundcontrol/src/comm/QGCMAVLink.h:20,
                 from ../qgroundcontrol/src/comm/LinkInterface.h:29,
                 from ../qgroundcontrol/src/comm/MAVLinkProtocol.h:29,
                 from ../qgroundcontrol/src/comm/MAVLinkProtocol.cc:28:
../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/../mavlink_helpers.h:68:25: note: declared here
 MAVLINK_HELPER uint16_t mavlink_finalize_message_chan(mavlink_message_t* msg, uint8_t system_id, uint8_t component_id, 
                         ^
../qgroundcontrol/src/comm/MAVLinkProtocol.cc: In member function 'void MAVLinkProtocol::_sendMessage(LinkInterface*, mavlink_message_t, quint8, quint8)':
../qgroundcontrol/src/comm/MAVLinkProtocol.cc:472:147: error: too many arguments to function 'uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)'
     mavlink_finalize_message_chan(&message, systemid, componentid, link->getMavlinkChannel(), message.len, message.len, messageKeys[message.msgid]);
                                                                                                                                                   ^
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/../protocol.h:85:0,
                 from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h:30,
                 from ../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/mavlink.h:25,
                 from ../qgroundcontrol/src/comm/QGCMAVLink.h:20,
                 from ../qgroundcontrol/src/comm/LinkInterface.h:29,
                 from ../qgroundcontrol/src/comm/MAVLinkProtocol.h:29,
                 from ../qgroundcontrol/src/comm/MAVLinkProtocol.cc:28:
../qgroundcontrol/libs/mavlink/include/mavlink/v1.0/ardupilotmega/../mavlink_helpers.h:68:25: note: declared here
 MAVLINK_HELPER uint16_t mavlink_finalize_message_chan(mavlink_message_t* msg, uint8_t system_id, uint8_t component_id, 
                         ^
make: *** [obj/MAVLinkProtocol.o] Error 1
19:19:25: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.5.1 GCC 64bit)
When executing step "Make"
```

TSC21, 2016-07-04:
Solved after updating the module directly on the mavlink folder with `git pull origin master`. So the hash must be updated on master.

closed 2016-07-04 by TSC21

NaterGator, 2016-07-04:
You could also try

`git submodule sync`

`git submodule update`