mavlink/qgroundcontrol #3725: Land abort button (MAV_CMD_DO_GO_AROUND)

dagar, 2016-07-04:
I'm implementing landing abort for PX4 fixed wing so I'd like to add a nice big abort button only when the vehicle is actually landing. 

How should I do this?
Unless I'm missing something obvious there's currently no way for the GCS to know if the mission is on on a landing waypoint?

DonLakeFlyer, 2016-07-05:
You'd need to plumb an isLanding into FirmwarePlugin. Then figure out some way to determine that (not sure how you do that). Then tweak the guided buttons to change based on isLanding. Also need top check for support for GO_AROUND in supportedCommands.

DonLakeFlyer, 2016-07-05:
Vehicle gets current mission item index updates. If the current mission item (you can look it up in MissionManager) is a landing one (not sure what that is) set an isLanding bool in the Vehicle?

dagar, 2016-07-05:
Isn't it possible we don't have the mission locally?

DonLakeFlyer, 2016-07-05:
We always have the mission locally that we are flying. That's how we show waypoints onthe Fly view. If we don't or it is out of date, that is a bug. Only way to be out of date (without a bug) would be is the Vehicle modified the mission on it's own. But that would be incorrect firmware I think.

dagar, 2016-07-05:
Alright that sounds acceptable for now. Maybe with mavlink 2 I could add an isLanding to EXTENDED_SYS_STATE.

Antiheavy, 2017-01-31:
Interesting topic.  I realize this is old, but I have a suggestion:

What if the [Abort] button is always shown on screen after takeoff, maybe with the other action buttons at the bottom of QGC.  Anytime the [Abort] button is pressed MAV_CMD_DO_GO_AROUND is sent and the aircraft will climb to a safe altitude while roll limited and then loiter above the Land waypoint. 

Would this show up as an "abort" mode type?  Is my understanding correct that any other command will exit the abort mode - e.g. pressing RTL again to go back and try another landing?  Should this be tied to having a valid Land waypoint?

cheers.

dagar, 2017-02-02:
Most of the the specifics of MAV_CMD_DO_GO_AROUND will be a PX4 issue. 
There's not a lot of detail in the mavlink spec. http://mavlink.org/messages/common (search for MAV_CMD_DO_GO_AROUND) and QGC is only responsible for sending the command (if we can find a way to display it at the right time).

dagar, 2017-06-26:
Implemented

closed 2017-06-26 by dagar