mavlink/qgroundcontrol #3727: The voltage detected by QGC lower than the battery actual value sometimes

glody, 2016-07-05:
Test Environment
QGroundControl: Daily Build: v2.9.7b-379-g2e9d112
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the voltage detected by QGC is correct
Step:
1. Lanuch QGC
2. Connect to Iris+
3. Check the voltage on the status bar
Excepted result:
The detected value should fit for the real battery voltage
Actual result:
The real voltage is higher than 11V, the detected value is lower than 10V sometimes.

[image]

DonLakeFlyer, 2016-07-07:
Is this different than what you would get with Mission Planner? We are using the exact parameter settings as downloaded from the ArduPilot site (as downloaded/set through Airframe config). If they don't quite work correctly the problem is more with the parameter settings than QGC.

DonLakeFlyer, 2016-07-07:
I also checked with my Iris+ and was getting readings that were faily close to an external reading. Maybe you need to reboot after download params for airframe so they will take affect?

DonLakeFlyer, 2016-11-10:
The voltage comes from the firmware side. So if the parameters are set correctly, which I have verified they are then there isn't anything QGC can do about this.

closed 2016-11-10 by DonLakeFlyer