mavlink/qgroundcontrol #3747: Code Signing

tops4u, 2016-07-08:
Please use Code Signing for further Releases of QGC for Mac as more malicious Software for Mac is emerging and it is anyoing to always Enter Admin to Start the Application first Time. Also Apple plans for the next release to rise the bar for unsigned Code to run in Mac OS.

dagar, 2016-07-08:
This also applies to the Windows build.

DonLakeFlyer, 2016-07-08:
@dogmaphobic iOS is signed right? So will OSX once it gets to the store as well right?

DonLakeFlyer, 2016-07-08:
Windows, iOS, OSX and Android Stable will all be available from the store, hence signed. Daily Builds of Windows and OSX will not be signed or be in the Store.

dogmaphobic, 2016-07-08:
As far as Mac OS is concerned, we only have a developer account (and signing certificate) for iOS. We will need to get one for Mac OS as well. Unlike iOS however, all builds can be signed and made available. It's just one extra step in the build process, so signing dailies is no big deal. The complicated part is setting up for the App Store, which dailies aren't going to any way.

NaterGator, 2017-12-29:
I don't see us in the Windows store... is that happening?

I have a code signing cert for my LLC, which I got through the MS provided discount on digicert issued code signing certs at 50% off (regular for $111/yr or EV for $224/yr). http://go.microsoft.com/fwlink/?LinkId=393249 

I'm happy to encrypt and push my cert for signing release builds but they'll show up as signed by NaterGator Engineering LLC.  Thoughts?

dagar, 2017-12-30:
We have a Windows Dev center account (check your gmail for an invite), but I haven't had time to look into actual package requirements to submit QGC. Should we go straight to a WinRT build or try to get the desktop version accepted? It would be great if you wanted to push this through.

DonLakeFlyer, 2017-12-30:
I'm going to take a look at this now that I'm wrapping up 3.3

closed 2024-06-01 by HTRamsey