mavlink/qgroundcontrol #3749: Radio Cal: Unusable over radio

DonLakeFlyer, 2016-07-08:
Seems like latest PX4 firmware has greatly lowered stream rate of RC channels message to QGC. This causes radio cal / flight mode setup to be unusable since the RC updates are too slow.

DonLakeFlyer, 2016-07-08:
@LorenzMeier I'm still doing some testing here. Did the rates change recently?

DonLakeFlyer, 2016-07-08:
Looking at firmware: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1861
Looks likes this has been at 1 per second over radio for a long time. Guess I haven't done a radio cal over radio link in ages. Either that or telling firmware were doing a radio cal used to bump stream rate and it doesn't any more.

DonLakeFlyer, 2016-07-08:
Two options (both on firmware side):
1) Bump standard stream rate to say 3 times a second. That should work.
2) Bump stream rates only during radio cal

dagar, 2016-07-08:
That's just the requested rate. Everything is scaled in proportion to the overall size of that mavlink mode (each message * rate) and the startup rate (1200). Then it only gets worse from there as the link degrades.

DonLakeFlyer, 2016-07-08:
Hmm, maybe I had a really crappy link. Let me try it again.

DonLakeFlyer, 2016-07-08:
Over WiFi I can get about 5Hz on RC_CHANNELS. Over SiK Radio I get 0.3Hz. This used to work ok in the past.

dagar, 2016-07-08:
Wifi on the pixracer? It's started differently. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L491

I wonder if the contents of MAVLINK_MODE_NORMAL have grown or if something regressed in the mavlink module. I'll try it on my setup this weekend.

DonLakeFlyer, 2016-07-08:
> Wifi on the pixracer?

Right.

DonLakeFlyer, 2016-07-09:
I just did the same test with ArduPilot and my Iris. Everything fine there, so my radios are working fine.

DonLakeFlyer, 2016-07-09:
I was able to replicate the same problem in my test bed setup. Still only .3Hz for RC_CHANNELS.

dagar, 2016-07-10:
It looks like @LorenzMeier improved it on master.

LorenzMeier, 2016-07-10:
Yep, please retry. It should be quite different now.

DonLakeFlyer, 2016-07-10:
Fixed.

closed 2016-07-10 by DonLakeFlyer