mavlink/qgroundcontrol #3751: CI: Setup for Stable branches and RCs

DonLakeFlyer, 2016-07-08:
Here is the process I'd like to use:
- Daily builds come from master
- Stable builds and RCs come from a branch
- When a release is ready for RC we create a Stable_#.#.# branch, no tag yet
- CI systems are updated to build this branch as well as master for all pulls and merges
- Once we are happy with RC, we tag the branch
- CI systems trigger on that tag and drop Stable builds
- Hotfixes go to branch, CI systems build, new tag created when happy with build

I think this is mostly doable on Travis. Don't know about AppVeyor. The real problem is CI Android daily builds during an RC phase. I'd like to have those switch to installing RCs but I'm not sure that's possible.

DonLakeFlyer, 2016-07-08:
@dagar Possible?

DonLakeFlyer, 2016-07-08:
FYI: The Windows and iOS Stable store uploads can still be manual for now

dagar, 2016-07-08:
Definitely doable. How do you want to version master? If the tag isn't off master then git can't walk back to get the last release tag (version number). What if development builds were just `development - date - gitsha`.

DonLakeFlyer, 2016-07-08:
Yeah, dev builds with date in them would be handy. Right, tag or Stable would be on branch, not master.

DonLakeFlyer, 2016-07-09:
@dagar I'd like to branch codebase now for Stable 3.0 (no tag yet). But first I want to make sure that doesn't screw up CI in any way. Also need to make sure we keep getting builds for pull requests against Stable 3.0 branch. There are still a couple issues out there which will require fixes but I want to start the process of working against a branch for Stable and open up master for 3.1. Do you have the time to fix this up?

dagar, 2016-07-09:
Go ahead and branch. You'll still get builds uploaded to s3. Replace master in the URL with branch name. Android beta builds and master versioning are the only issues I can think of at the moment.

I don't know what the best solution is for Android. For now we could have the Google play release track set in the travis config. master = alpha, Stable_3.0 = beta, tags = production

We could just switch versioning to be branch-sha-data, or tag. 

Does that cover everything?

DonLakeFlyer, 2016-07-09:
I think for android we just have Daily Builds and Stable. We relable what we were doing as Beta to be Daily Build. There is no drop to Android for builds on Stable branch which are not tagged.

dagar, 2016-07-09:
That certainly makes it easier. It would only matter if you did much in the Stable branch before release while master development continued.

dagar, 2016-07-09:
With the development versioning updated here https://github.com/mavlink/qgroundcontrol/pull/3762 and the build deploys adjusted/fixed we should be good to go. I'll keep a close eye on the first one to make sure everything works as intended.

DonLakeFlyer, 2016-07-10:
Thanks Daniel. Waiting to branch since I just found another problem.

DonLakeFlyer, 2016-11-10:
This is mostly all working now. Still no automatic iOS but that's a much bigger problem.

closed 2016-11-10 by DonLakeFlyer

dagar, 2016-11-11:
@DonLakeFlyer @dogmaphobic @LorenzMeier 
I've been using https://buddybuild.com/ for other iOS and Android projects. It's yet another continuous integration service, but focused on mobile. The main benefit for us is they handle all signing issues and deployment to both beta and production tracks of each store (including TestFlight). We just need to find a way to use the commercial version of Qt.