mavlink/qgroundcontrol #3757: Map tiles - use already available tiles resized while fetching proper tile

dagar, 2016-07-09:
This would actually feel like a big improvement during regular use.

I think this has been discussed before, but I couldn't find an issue.

dogmaphobic, 2016-07-09:
This cannot be done unless you want to rewrite the internal map engine in Qt itself. Qt handles mapping. It asks our "plugin" to provide a tile for a given grid index and we supply it (if it exists). The rendering and everything else is done within Qt. If I were to reply with a resized tile, we would not be able to replace it with the real one once its fetching is complete.

There are a few issues related to tiles (#2368, #3111, etc.), which all are hampered due to the same issue above.

dagar, 2016-07-09:
That's unfortunate.

dagar, 2016-07-10:
@dogmaphobic 
I poked around the Qt source and found this in QGeoTileFetcher.

    public Q_SLOTS:
        void updateTileRequests(const QSet<QGeoTileSpec> &tilesAdded, const QSet<QGeoTileSpec> &tilesRemoved);

What if QGeoTiledMapReplyQGC::cacheError set the tile image to the closest available tile already cache (resized), and kicked off the proper tile download somewhere else. The downloaded tile could go straight into the cache and then call updateTileRequests() for that TileSpec.

What do you think?

dogmaphobic, 2016-07-10:
It's not quite the way things work. If you return an error for a tile, that tile will never be requested again. If you return a "enlarged" tile, there is no simple way to tell the Qt map engine to reload that tile. `QGeoTileFetcher::updateTileRequests` is an internal Qt function based on the internal state of the tiles. I guess we could try calling it and see what happens but my gut feeling is that it would mess up Qt's internal tile states. I just have not had the time to spend with these "tweaks".

DonLakeFlyer, 2016-09-18:
Closing this since likely not doable

closed 2016-09-18 by DonLakeFlyer