mavlink/qgroundcontrol #3763: Log Download: Refresh never shows list

DonLakeFlyer, 2016-07-10:
Log entries come through but are never processed because _requestingLogEntries is false at this point:
https://github.com/mavlink/qgroundcontrol/blob/master/src/ViewWidgets/LogDownloadController.cc#L151

DonLakeFlyer, 2016-07-10:
Looking at this a little more it looks like the timer fires, which then leads to this call tree.
0	LogDownloadController::_receivedAllEntries()	LogDownloadController.cc	236	0x10028f880	
1	LogDownloadController::_findMissingEntries()	LogDownloadController.cc	289	0x10028e698	
2	LogDownloadController::_processDownload()	LogDownloadController.cc	122	0x10028e3ca	

This happens before the first log entry comes back so the list is empty and _receivedAllEntries marks the request as complete.

dogmaphobic, 2016-07-10:
Odd, I wonder how it worked until now. I'll take a look in the morning.

DonLakeFlyer, 2016-07-10:
Timing thing? Timer goes off before first entry comes back. I have a PixRacer with a pile of logs on it. Maybe the big list causes the initial delay of the first entry.

DonLakeFlyer, 2016-07-10:
Tomorrow I'll try on a different board with fewer logs.

DonLakeFlyer, 2016-07-10:
I works fine with smaller log counts.

DonLakeFlyer, 2016-07-10:
When I do an Erase All it erases all the files in the log directory but it does not erase the msg_*.txt files. These pile up as well for each run and could fill the SD card causing the vehicle to crap out.

closed 2016-07-26 by DonLakeFlyer