mavlink/qgroundcontrol #3767: Feedback:File formats make QGS adoption harder.

AndKe, 2016-07-10:
.mission files do not stick to the official mavlink format: http://qgroundcontrol.org/mavlink/waypoint_protocol#waypoint_file_format

lots of GCS use this format, moving to QGS will require me to convert all missions (annual repeating mapping missions) to this format , every season some refinement is done to the missions, so I also keep a history of such missions. 

Beside switching to something that can be called a proprietary format (that only one GCS uses)
Another serious reason to that we may generate QGC WPL files using other tools, and/or process such files for altitude verification in mountain /valley areas.

So I feel strongly that QGS should offer to save/load the official format as well.

Second issue:  .params file format.
the params format, unnecessary complicated, starts with MAVID, which is, after all - a parameter.
I understand the attempt to be able to save multiple COMPONENT_ID's to one file, but it does not provide any real advantage beyond offering to save subcomponents to separate files.
Again: it would smooth the transition to be able to save to a simpler format like AP2 , missionPlanner, and many other use.

DonLakeFlyer, 2016-07-10:
QGC supports all old formats as input. It only writes in the new format. QGC requires a new format for missions since it has the concept of a Complex Mission Item. No other GCS has this concept (except may be Tower). A Complex Mission Item encpsulates multiple waypoints as well as additional information into a single item. This allows you to save a mission and then re-edit the Complex Mission Item as the single entity again.

A simple example of what this allows you to do with Survey:
- Create a survey, save it to a file
- Go the the field, re-open the Survey
- Adjust the grid angle (or anything else) to wind conditions
- Send to Vehicle

When you reload the Survey it hasn't degraded into a massive pile of waypoints. It continues to be a Survey. You can try how this works in the 3.0 release. We are working on two Complex Mission Item types for 3.1: Survey and Structure Scan.

Bill Bonney (AP2 lead dev) proposed the initial .json format that I started to work with. The reason being that not only the above is an issue. But even though the file formats used to be the same you couldn't actually interchange files between QGC and AP2/MP. This is because of ArduPilot handling of waypoint 0 as a fake home position. PX4 stack doesn't not do. If all you have is an old file, you have no idea how to handle waypoint 0 given the firmware you are running. The new file format handles this by explicity calling out a "Planned Home Position" waypoint.

We'll look at supporting write to the old format but it will not be able to support the new features in QGC. And you will be back to the home position problem.

As far as I know the parameter file save forrmat is exactly the same as AP2 and MP. It hasn't been changed in years. I'll take a look to see what is different.

AndKe, 2016-07-10:
Yes, I do understand the need for a method to organize WP's , for re-editing survey's and such.
(always hoped for a method like this:)
wp1
wp2
survey start , 
survey polygon
survey polygon
survey polygon
survey polygon
wp3
wp4
wp5 (these are "normal" WP's , generated by "survey" data)
survey END
wp6
wp7...

This way - a survey could be edited , by replacing all WP between start, based on survey ppolygons, praameters, etc...   but the WP's itself, would be absolute - static until ergenerated - verifyable, and in the same place even when opening this flightplan after 2 years with new version.

The scary part here, is that is a part of survey needs different elevation, there is no way to do it.
Another scary part, is that loading this mission in the future, may calculate it entirely differently..

As for the ,param vs .params  .. there is huge difference:
.param
AHRS_TRIM_Y,0.00912481
AHRS_TRIM_Z,0.00000000
AHRS_WIND_MAX,0

.params
20	1	AHRS_TRIM_Y	0.009124812670052052	9
20	1	AHRS_TRIM_Z	0.000000000000000000	9
20	1	AHRS_WIND_MAX	0	2

DonLakeFlyer, 2016-07-10:
If you are looking at a saved Survey mission in 3.0 you won't see the raw waypoints for the Survey. That is just a fluke of work in progress. The format is meant to be readable by GCS which do not understand the Complex Mission Item concept, that is GCS specific and taggedd with the GCS name. I had it output the full list previously so any GCS could at a minimun can load all waypoints, but I was having problems with it so I took it out. It will come back. This means that any GCS can read the .json and get the exploded versions of any Complex Mission Items.

Not quite sure what to do about the param format thing. Have to think about that.

DonLakeFlyer, 2016-07-10:
I was supposed to have picked up the .json file format discussion with Bill (AP2) and Michael (MP) after Bill and I had talked about it a while back. But I forgot. My screwup.

DonLakeFlyer, 2016-07-12:
FYI: Mission Planner now supports new .json mission format. AP2 coming soon.

Still haven't discussed param file format.

AndKe, 2016-11-04:
Related news: "Tower" dropped .dpwp mission format and went back to ".txt" (QGC WPL) - for "better compatibility".   - It may be smart to have an option to export to txt as well.

DonLakeFlyer, 2016-12-31:
I'll add param read support for MP format and output to old version mission format in 3.2

AndKe, 2017-01-02:
by any change , is the "MP format" the QGC QML  ? (used by mavproxy and many more ?)  - or is it yet another thing ?

DonLakeFlyer, 2017-01-02:
MP and QGC mission text (non-json) formats are the same. QGC can read text format, but can't write text format. MP supports read of V1 Json format, only writes text format.

DonLakeFlyer, 2018-06-20:
There is a separate issues for saving to text file formats

closed 2018-06-20 by DonLakeFlyer