mavlink/qgroundcontrol #3773: Please save/restore map position on exit/load.

AndKe, 2016-07-10:
Yes, I do recognize the park near the golden gate bridge where I hired bicycle to go over the bridge :)  - still, I usually plan missions in northern Norway, and it's much better not to need to pan across the world every time just because the park is a nice place :)

DonLakeFlyer, 2016-07-10:
The map will go back to the last vehicle home position that it got. If you've never connected a vehicle it will default to SF. Does that make sense?

AndKe, 2016-08-02:
Ok, I understand how it works.
Still , please save/restore last position.
I would like to add maps for Norway/nordic countries. (very useful, topographic and detailed maps)
If such maps were enabled, and application started, showing USA as usual, the map would be empty.
Also, it's very konveniens to spawn the map of the area where you live/plan/fly missions, instead of half-a-globe away :)

DonLakeFlyer, 2016-12-24:
I'm going to revisit all of this in 3.2

DonLakeFlyer, 2017-04-06:
All reworked now

closed 2017-04-06 by DonLakeFlyer