mavlink/qgroundcontrol #3777: Load custom command QML file failed

glody, 2016-07-11:
Test Environment
QGroundControl: Daily Build: v2.9.7b-411-g0fefa4f
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the info view window shows normally
Step:
1. Lanuch QGC
2. Connect to SITL
3. Open "Custom Command" from "Widgets" menu
4. Save http://qgroundcontrol.org/custom_command_qml_widgets as a qml file
5. Click "Load Custom Qml file..." in Custom Command window and select the qml file saved in step4
Excepted result:
The qml file is able to load
Actual result:
"Parameters missing from firmware: -1:MAV_SYS_ID." shows

glody, 2016-07-11:
[image]

DonLakeFlyer, 2016-07-11:
This example code is using a parameter from the PX4 flight stack and you are running ArduPilot so that param does not exist. QGC is complaining correctly. When I re-do this part of the docs I'll try to point that out.

closed 2016-07-11 by DonLakeFlyer