mavlink/qgroundcontrol #3778: Some offline map size is 0

glody, 2016-07-11:
Test Environment
QGroundControl: Daily Build: v2.9.7b-411-g0fefa4f
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the offline map size is correct
Step:
1. Lanuch QGC
2. Click "Add new set" in Offline Maps item Settings page
3. Download the set
Excepted result:
The map size should not 0
Actual result:
Some offline map size is 0
[image]
[image]

dogmaphobic, 2016-07-11:
I not only was able to reproduce this, I also found a few other oddities with the display of sizes and download status. Not sure what happened as this was all working at some point. Looking...

glody, 2016-07-26:
Checked on daily build HEAD:e6e0429 2016-07-20 18:58:20 -0400, this issue unable to reproduce. close it. thanks.

closed 2016-07-26 by glody