mavlink/qgroundcontrol #3779: Surveying parameter from camera parameters

birchera, 2016-07-11:
We would like to implement the direct computation of surveying parameters (altitude, grid spacing) from camera parameters (sensor size, focal length, ...). Additional required inputs are ground sampling distance (GSD) and overlaps in both directions.

For the camera parameters we propose to add fields to the camera settings tab like this:

[image]

We have thought of three different ways to include this functionality in the survey widget.
* Adding a button to the widget and select GSD and overlaps in a pop-up window and then overwriting altitude and grid spacing.
[image]
* Checkbox in widget, when checked reveals the fields for GSD and overlaps.
<div class="nav3" style="height:705px;">
[image]
[image]
</div>
* Adding GSD and overlaps to the camera settings tab and give computed values as defaults when creating the survey item.

@DonLakeFlyer does this feature fit your plans for the surveying and do you have a preference for one or the other?

DonLakeFlyer, 2016-07-12:
The first step is to add support for the base functionality and then figure out the right UI for it later. So for now that means all the tweakable values just go into the Survey item editor.  So the first step is to list out the changeable values that you want. The best place for that is in the main Survey discussion issue: https://github.com/mavlink/qgroundcontrol/issues/2507. Then we can home some discussion around those. If there aren't any issues and I easily add to Survery editor and then the implementation can begin.

birchera, 2016-09-02:
This has been adressed in https://github.com/mavlink/qgroundcontrol/pull/3981.

closed 2016-09-02 by birchera