mavlink/qgroundcontrol #3783: qgc shows up blank on second monitor

clovett, 2016-07-11:
Windows 10, 3 monitors, qgc remembers which monitor it was on.  When it tries to show up on my non-primary monitor it shows up completely blank (white).  I have to move it back to primary monitor, restart it, then move it back to get it working again.

NaterGator, 2016-07-11:
Which version are you running? I am familiar with this and recall it being an old Qt bug about screen management.

clovett, 2016-07-11:
2.9.7b.254, which seems to be the latest... I haven't tried the v3 rc builds...

NaterGator, 2016-07-12:
Hmm, what type of computer, graphics device, and graphics driver are you running?

Does starting QGroundControl in compatibility or safe mode alleviate the issue?

clovett, 2016-07-12:
It's an HP z420 desktop with AMD FirePro W7000, with ATI drivers dated 11/29/2015, version 15.3.
I don't need to use safe mode - all I need to do is move the window back to my primary monitor and restart the app.  It's when it launches to a secondary monitor that it has problems.  But safe mode also fixes it because safe mode always goes back to primary monitor.

clovett, 2016-07-12:
Note: I do not have this problem at home with same version of windows 10 and NVIDIA GeForce GTX 650.  So I see that this behavior is dependent on the graphics card.

dogmaphobic, 2016-07-12:
For what's worth, I run with 3 monitors and I have never seen this (Mac Pro).

NaterGator, 2016-07-12:
@clovett just to avoid confusion: I mean running QGC from the safe mode link in the start menu, not Windows safe mode. If you can get compatibility mode to load on the correct screen too that would be informative.

clovett, 2016-07-12:
Yes, I found the QGC safe mode, so that's what I assumed you were talking about.  I'll see if I can get safe mode to load on the secondary screen, but I recall it would reset the screen it had saved and go back to the primary screen.

DonLakeFlyer, 2017-04-13:
Any status update on this?

lovettchris, 2017-04-17:
seems to be fixed now, haven't seen this problem in a while, on two different multimonitor PC's running windows 10.

closed 2017-04-17 by clovett