mavlink/qgroundcontrol #3788: Can't download latest QGC for linux

CarlOlsson, 2016-07-12:
If I press **Linux tar download** on http://qgroundcontrol.org/downloads I get redirected to:
[image]

dagar, 2016-07-12:
In qmake the app was renamed qgroundcontrol -> QGroundControl (to fix android naming), which also updated the installer ouput.

https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.AppImage
https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.tar.bz2

I'll get the wiki updated. Thanks for reporting.

dagar, 2016-07-12:
It looks like I don't even need to update the wiki. Downloads is taking me to the new gitbook which has the proper links.
https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html

closed 2016-07-12 by dagar