mavlink/qgroundcontrol #3797: Survey Phase 1 Implementation [Discussion]

DonLakeFlyer, 2016-07-12:
This is the discussion area for phase 1 implementation.

DonLakeFlyer, 2016-07-12:
* Polygonal area which defines the survey area

DonLakeFlyer, 2016-07-12:
* Ability to have multiple survey areas in a single mission

DonLakeFlyer, 2016-07-12:
* Ability to re-edit survey after save/re-load from file (doesn't degenerate into big pile of waypoints)

DonLakeFlyer, 2016-07-12:
Values needed to generate a simple grid flight pattern over the polygon

DonLakeFlyer, 2016-07-12:
Values needed to fly a correct turnaround to the next grid line

DonLakeFlyer, 2016-07-12:
Conversion of above information into a set of waypoints

DonLakeFlyer, 2016-07-12:
Load/Save to file of a mission which includes a survey

DonLakeFlyer, 2016-07-12:
**Feature: Survey is treated as a single entity of the overall mission plan**

WickedShell, 2016-07-13:
If like to have a discussion about that JSON format. I'm looking to support it as well, however there are a couple of really weird aspects to it last time I looked at the output that if like to discuss changing before everyone has locked into it. Would that be held here or maybe on the drone code discourse instead?

WickedShell, 2016-07-13:
Sorry for duplicate comments, web interface on a phone is apparently poor and I can't delete it..

DonLakeFlyer, 2016-07-13:
> If like to have a discussion about that JSON format.

Yup. I'm going to break that out into a separate discussion and bring in Bill and Michael as well. I'll get that organized hopefully by tomorrow.

DonLakeFlyer, 2016-07-13:
Also if your looking at an example saved mission file with a survey in it on daily builds that format is wrong. It doesn't write out all the raw waypoints.

DonLakeFlyer, 2016-07-13:
Added separate discussion for JSON file format: #3800

birchera, 2016-07-13:
We have been testing the surveying feature for a while now and have had some good experiences with it. 
The features listed here address most of the concerns we had with surveying so far and it's great to hear that you plan to work on them in the near future.
Being able to resize / move the survey area by moving existing polygon points, better/more customized handling of turnarounds and having the survey item with the corresponding meta data and all the individual waypoints in the mission file, would make using the surveying feature much easier for us.

An additional helpful feature would be to output the total flight distance/time of the survey, but this might already fall into a later phase.

We are happy to test any of the new features.

DonLakeFlyer, 2016-07-13:
Added separate discussion for turn-around: #3801

DonLakeFlyer, 2016-07-13:
Feature: Mission stats

This feature is to calculate a set of information associated with the mission and display to the user. For example total flight time and distance, battery usage and so forth.

* Stats discussion is here: #3802

DonLakeFlyer, 2016-07-13:
Added new feature "Mission Stats" to Phase 1 and separate discussion #3802 for it.

DonLakeFlyer, 2016-07-24:
Adjustment of existing survey polygon now supported.

DonLakeFlyer, 2016-07-26:
**Feature: Ability for user to specify information which determines entry into survey area**

This feature allows the user to determine how and where the vehicle enters the survey grid.

* Discussion here: #3864

DonLakeFlyer, 2016-07-26:
Added bug #3866

DonLakeFlyer, 2016-08-23:
Add camera discussion: #3963

DonLakeFlyer, 2017-04-13:
This is mostly all done now

closed 2017-04-13 by DonLakeFlyer

SpandanKumarSahu, 2017-12-12:
@DonLakeFlyer, I am new to QGroundControl. I was trying to figure out, if along with storing the survey, we can have an option of storing the generated waypoints too?

DonLakeFlyer, 2017-12-12:
At some point I'll do that, but the real question is why? Even if they were there you would not be able to modify them. The only reason to do it is for other ground stations to be able to read the format. If you want to get the waypoints for a survey you can just round trip the plan through the vehicle:
* Upload
* Download - It will denegenerate into individual waypoints
* Save

SpandanKumarSahu, 2017-12-12:
Yeah, I understand it that might not be a very relevant feature. Thanks for the reply, anyway.