mavlink/qgroundcontrol #3799: Survey: Simple grid visualization discussion

DonLakeFlyer, 2016-07-12:
Screen shot shows the simple grid display as well as entry exit point display:
[image]

DonLakeFlyer, 2016-09-02:
I wonder if the grid angle should automatically set itself to set up the grid lines such that they are following the longest path.

kellysteich, 2016-09-05:
It would be great to have the option of automatically aligning the grid lines such that they are following the longest path.
You would most likely still need to give the user the option of aligning the grid lines with the wind direction, though. Either also have an option of doing this automatically or still leave the option of setting the grid angle manually.

DonLakeFlyer, 2017-04-13:
I think this is good as it is

closed 2017-04-13 by DonLakeFlyer