mavlink/qgroundcontrol #3800: Mission: JSON file format

DonLakeFlyer, 2016-07-13:
Discussion for mission JSON file format

DonLakeFlyer, 2016-07-13:
**The complexItems field**

The concept behind the **complexItems** field is to provide a location to store the meta data associated with a complex mission items. You can have multiple complex mission items within a single mission. Examples of complex mission items are Survey and Structure Scan. A single complex mission item stores meta data associated with the item. It references a range of mission items from the main **items** list.  The mission items associated with a complex mission item are not stored within the complex mission item. This storage structure allows for two things:
1) Round tripping a re-editable complex mission item through a file.
2) Ground stations which don't understand complex mission items still have access to the full raw set of mission items.

WickedShell, 2016-07-13:
In an airport at the moment will have to review again later but the first comment that comes to mind is this looks difficult to reinsert the complex items into the simple items (ie a mission of take off then survey then some manually planned waypoints)

Why not put all items in a single list, and require that a complex item also provide a list of all it's generated simple items that everything can parse? On first pass this sounds slightly easier to load fort the gcs that can interpret the complex items.

WickedShell, 2016-07-13:
Version of the file format is valuable but I could definitely see wanting to support logging the gcs version as well for compatibility reasons

WickedShell, 2016-07-13:
A comment on the camera metadata for survey is that any non trivial/professional  gcs will be needing to store the camera parameters rather then the offsets of the lines in meters. But maybe it doesn't make sense to do a common variant of this complex item even though it would be nice

DonLakeFlyer, 2016-07-13:
> Why not put all items in a single list, and require that a complex item also provide a list of all it's generated simple items that everything can parse? On first pass this sounds slightly easier to load fort the gcs that can interpret the complex items.

I've kind of gone back and forth on that exact thing. It all got so complicated to code, current QGC doesn't even write out the complex item mission items to the file anywhere because it got crazy trying to manage the referencing. I don't think add the items to the complex item as well will make things easier. The reason is that you still have to handle the items coming back and forth from the simple items to complex items and back to simple items. So you still need to code to skip over items in the main item list. In reality not quite sure. Need to think about it a little.

WickedShell, 2016-07-13:
I guess what I want sure is I didn't see how to slot the complex item back
into the simple list correctly... ie waypoints 5-100 are all from complex
item one but I couldn't see how to cut that back in.

Particularly if you are loading a file where the number of generated
waypoints for the complex item is now different due to either but fixing
bugs or new attributes that weren't present before causing a change in the
number of waypoints.

DonLakeFlyer, 2016-07-13:
> any non trivial/professional gcs will be needing to store the camera parameters rather then the offsets of the lines in meters.

This is the current level of survey support in QGC showing through somewhat. It doesn't currently support any camera information, so all you have is grid distance. In the long run the camera information, plus overlap and so forth in turn sets the grid distance. A question is whether it ends up being an either or thing. If you have camera information you have just that. But if you don't you can still create a simple grid using just grid distance without needing to know anything about camera stuff. Use case which comes to mine happens to my RC field quite a bit. Send out a drone to go look for a lost plant. The person that does that probably doesn't know anything about the details of camera or more likely video camera they have on their drone. Just just set up a simple grid by distance.

WickedShell, 2016-07-13:
From the professional mapping perspective the camera information and
handling is mandatory, and if the user can't get the camera info then they
have no chance of guaranteeing good data anyways. But as you observed for
now this could be unique to each gcs.

DonLakeFlyer, 2016-07-13:
> I guess what I want sure is I didn't see how to slot the complex item back
> into the simple list correctly.

The complex item should have meta data something like this:
"startId": #
"endId": #

Which would indicate the range of ids from the main list which are for the complex item. I would expect the GCS that supports the complex item to simply skip loading these from the main list and regenerate them from the meta data. That is how QGC currently works (especially since it doesn't even write the complex items!).

WickedShell, 2016-07-13:
It just seems like a slightly weird load process to me add you have to scan
the complex before filling the gaps from the simple. Seems like just
iterating through a combined list and loading a simple subset for any
complex you don't understand is easier. But that's just my thoughts without
having sat down to implement it yet.

DonLakeFlyer, 2016-07-13:
I think the issue is that no matter how you structure it, if you want to make downlevel gcs loading support feasible/simple, then you move the pain to the up level gcs that understands complex items. Maybe some completely different structure or additional side structure is possible to simplify things. I'll think some more about it in the next couple days. But I know exactly what you are talking about since I already ran head first into it with current QGC code.

NaterGator, 2016-07-13:
@DonLakeFlyer maybe in a JSON format you have both? Insert the complex and simple items, but tag the simple items with an ignored field that lower feature tier GCS software will ignore (like: "ephemeral": true) but high tiered knows to interpret. I guess the trick is tagging in the complex items so that it doesn't trip up lower tiered GCS packages, but I think there are clean ways around that.

billbonney, 2016-07-13:
This was the example I wrote before https://gist.github.com/billbonney/8d8b8798ba5bb7a50fa8

Where you would have support for mission_items, and then in sequence, mission_area, that defined a  polygon area with properties. those properties could be as simple as grid discretion, and spacing, or camera properties etc... 

I think having complexItems and missionItems as separate lists makes the format much harder to read and parse. 

We can add new complex items like initiate_landing take_off etc.. whatever, but we should try and document these and make them 'generic' for all gcs implementation to support.

Having the meta data about the program that writes the file may help resolve problems, but it shouldn't be required at all. If QGC has different complexItems vs. Mission Planner or Tower. It kind of defeats the purpose.

I know the main problem of implementing this currently in APM Planner 2.0 and MAVPilot is that the internal structure represents a mission_item in MAVLink, when really it should be an object list that represents the waypoints and behaviour on the screen. Only when it 'externalizes' to the autopilot does it get converted into mission_items. (And don't forget that there is also the case where the son could be uploaded to a onboard computer and it just guides the autopilot in real-time, point by point.)

Anyway, i'm glad this is being discussed 👍

DonLakeFlyer, 2016-07-13:
> I think having complexItems and missionItems as separate lists makes the format much harder to read and parse.

I think Bill hits the nail on the head. I was trying too hard to preserve a very simple format for downstream groundstations which don't understand a complex item. But since those gcs need to implement new JSON parse anyway it can be at least a little more structured. I like Bill's single list way of doing it much better. It's still easy for a gcs which does not support complex to read the raw items. I'll update the proposal at the top for that style format instead and we can take a look at it.

WickedShell, 2016-07-18:
I think I'm missing something but with Bills version where does a GCS that doesn't understand how to generate that survey grid get a list of waypoints? Expecting multiple GCS's to generate the same way points seems extremely unlikely/farfetched, as well as there being a lot of value in maintaining the information for a non compatible GCS to still extract and upload the data.

billbonney, 2016-07-18:
@WickedShell I wasn't looking at that use case. There could be an 'absolute waypoint only export'  feature that the GCS could implement. Or you could create an embedded abs waypoints array in the JSON file for interoperability.

see https://gist.github.com/billbonney/8d8b8798ba5bb7a50fa8

DonLakeFlyer, 2016-07-18:
> I think I'm missing something but with Bills version where does a GCS that doesn't understand how to generate that survey grid get a list of waypoints?

I'll make an example tonight. But basically what Bill showed.

DonLakeFlyer, 2016-07-19:
Here is a possible way to flatten out the top level structure and simplify reading for both gcs that support complex items and those that do not. This creates a linear read sequence for both only simple items and a read that supports both simple and complex.

The top level format is the same except that all items both simple and complex are in the **items** array.
```
{
    "MAV_AUTOPILOT": 12,
    "groundStation": "QGroundControl",
    "items": [ { simple or complex item }, { simple or complex item } ],
    "plannedHomePosition": { mission item }, 
    "version": "1.0"
}
```

The simple item pretty much stays the same as above, but the **type** field is now required.
```
{
    "type": "simpleItem",
    ...
}
```

The complex item is mostly the same as well except for **type** and the **items** array which contains the simple items associated with the complex item.
```
{
    "type": "complexItem",
    "items": [ { simple item }, { simple item } ],
    ...
}
```

So a downlevel gcs reads items linearly. When it see a complexItem it just reads the simple items from the complex item and pops back up to continue reading in the main **items** array. A gcs that support complex items just reads the complex item meta data and regenerates the simple items using its own algorithms.

WickedShell, 2016-08-01:
@DonLakeFlyer I prefer that a lot more over the two list approach! Seems much more straightforward for reading :)

How about adding rallypoints to a file as well? Would they just go under a seperate rally-items list? I know not all missions have support for them and that some will treat them differently, but it would be nice to be able to include this as part of a mission, without having to be a GCS specific bit.

DonLakeFlyer, 2017-01-02:
Current state of Mission/Survey file format is documented in dev guide: https://donlakeflyer.gitbooks.io/qgroundcontrol-developers-guide/content/MissionFileFormat.html

Rally Points/GeoFence are currently separate files. Further down the road I have the concept of a "Plan" file which could include mission, rally and geo-fence.

Also: Survey still doesn't output internal waypoints for downstream client use. Although the new V2 mission file format with respect to how it deals with complex items is now set up to support that.

DonLakeFlyer, 2017-04-13:
At this pint I'm not going to worry about downstream client reading

closed 2017-12-29 by DonLakeFlyer