mavlink/qgroundcontrol #3801: Survey flight path turn around

DonLakeFlyer, 2016-07-13:


DonLakeFlyer, 2016-07-13:
From @Andke
 Hi, I have long experience in mapping missions, often in narrow valleys, and over very varying altitude.
The planes in use, are very aggressively tuned (high roll angle) to turn efficiently in terms of space.
Here are a few tips for better planning, that I would love to see a tool use:
[image]

Please notice the small(normal) acceptance radius in WP1,2,3  - observe the way plane turns, then flies towards WP3, then turn again.

Now see that when it reaches WP4, WP5's acceptance radius is big (actually touch WP4 - resulting in a smooth single turn between and WP5)   -   resulting in better turn, and nice aligning with next path. 
The trick is, basically, that the acceptance radius of WP5,7 should be so big, it actually touches WP4,6  - This way the nav controller will align ASAP, not turn-level-turn.

Also, please allow to "inspect" the survey - see each WP , and set/modify each WP individually.

DonLakeFlyer, 2016-07-13:
Some question:
* I would expect that turn-around needed for fixed wing and turn-around for multi-rotor is different. Is that correct and necessary?
* Is the @AndKe example above only one possible way of doing a good turn-around? Or are there multiple mechanisns needs that the user choices from.
* In implementing @AndKe style turn-around what variables does the user need to provide in order to generate?

AndKe, 2016-07-13:
The only "variable" here is the UAV's turn radius, (but it does not need to be entered manually) 
- if for example the grid spacing is 50meters, and a big UAV of 6m wingspan requires 150m to turn, - then the operator already knows that - and adds another >100 to the area being mapped.

In related experience; when we are using bigger machines with bigger turn radius, then we are usually flying higher anyway (have wider space between grid lines.)

I think we all always draw bigger plans than the actual area that have been ordered, that too solves the "problem" with bigger or longer turns (usually due to strong wind - in cases where turning downwind cannot be avoided). 

On the second hand - it could be nice to let the user adjust the "extension" of WP 4,5 and so on , it's not necessary to do it for both.  Maybe WP4 type of wp could be defined as "extend by 20m"  - then WP5-type would move relatively to that. ?

birchera, 2016-07-13:
I'd also like to share our experience for surveying with our VTOL:
* I agree with @AndKe that other than the grid spacing it should only depend on the turn radius (for fixed-wings).
* We tried different versions with additional waypoints to shape the turnaround. What we found to be beneficial is an extension of the sweeps to account for the acceptance radius of the waypoint, as well as some lead distance for the aircraft to settle the tracking.

In general I don't see the drawing of larger polygons as a desirable solution. I'd rather put waypoints beyond it and let the user inspect them to be ok.

DonLakeFlyer, 2016-07-19:
Using the @AndKe drawing for reference. Specifically talking about WP4+5 which are on one side of the grid, and then WP6 which is on the other side of the grid. Also making the assumption that the user has specified the grid to exactly match their survey area.  Also this is for fixed wing. Some questions:
1) It would seem like WP4 needs to be outside the grid by the amount of the acceptance radius such that the vehicle doesn't start turning until it has complexly finished the pass.
2) After WP5 doesn't there need to be a WP6 in a similar position to WP4 (same side of grid) just on the new line such that the vehicle will line itself up correctly parallel to the new grid line?  Otherwise it will come out of WP5 and head to the other side of the grid on the angle that it left the acceptance radius?
3) Do we need to handle a case where the turn radius of the vehicle is larger than the grid spacing?

AndKe, 2016-07-20:
Yes, a batter way to put it , is that the WP with bigger radius may use grid spacing as radius, - and, be moved outside the target area with the distance of the radius.

As for larger vehicles, like I mentioned before, it's very common to draw plenty bigger pattern than what actually needs mapping, simply due to endurance.

Also - in my experience, any mapping mission is drawn much bigger than the actual area of interest, for good photogrammetry or laser the best elevation data is in areas with 9 or more overlapping photos, and more points (in case of laser) - so I usually plan all the outer passes to be outside the area to be mapped.   (thus always making bigger plan than really needed).

This is important; I do not expect the software to extend the area for me.  - It's important that I actually *see* where the plane is expected to go, and turn.  Because the mission may be close to mountains or huge high voltage transmission lines at altitude (hanging across a valley)

AndKe, 2016-07-20:
additional issue for similar things to have in mind related to the same code: https://github.com/mavlink/qgroundcontrol/issues/3827

DonLakeFlyer, 2017-04-13:
We don't have separate turn in/out values. But I think what we have now is good for 3.2

closed 2017-12-29 by DonLakeFlyer