mavlink/qgroundcontrol #3802: Survey: Mission stats

DonLakeFlyer, 2016-07-13:
Although not solely a feature for Survey generating a set of stats associated with the mission such as flight time and distance, battery usage and so forth is important. Let's use this issue to discuss what information should be collected about a mission.

DonLakeFlyer, 2016-07-13:
I'm in the process of collecting up the information from the various issues already entered on this...

birchera, 2016-07-13:
Related to https://github.com/mavlink/qgroundcontrol/issues/3772

In my experience the values that are interesting to know when planning missions are:
* Total flight distance
* Total time
* Max telemetry distance

For VTOLs:
* Splits hover/cruise distance
* Splits hover/cruise time

For Survey missions:
* Number of images that will be taken
* Covered area

@kellysteich FYI

DonLakeFlyer, 2016-07-16:
@andke In #3772 You have a speed value which I don't quite understand. Can you explain more?

DonLakeFlyer, 2016-07-16:
FYI to folks looking: @birchera and friends are actively working on support for this.

tubeme, 2016-09-11:
Another important parameter set is the Altitude information:

Higest altitude
Lowest altitude
Avarage altitude
Cumulative ascent
Cumulative descent

This could orient the pilot about the performance needed for the mission in terms of energy requirements.

DonLakeFlyer, 2017-04-13:
This is in a good state for 3.2

closed 2017-12-29 by DonLakeFlyer