mavlink/qgroundcontrol #3806: Off-Topic: Unable to download daily builds from AWS

shadow1runner, 2016-07-15:
I know that's not related to code, but I hope that that's not the completely wrong place for this.
Anyway, downloading [the latest Win build](https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/qgroundcontrol-installer.exe) gives the following error:
```xml
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>55D863879D9C3CC2</RequestId>
<HostId>
G9Z6ZSoRXv91sgpWO7zycWK7JpiJg2aZ7VJEShfaEDU5KDSU91LygxmnO3tXIqfAkkLAGFkaqmY=
</HostId>
</Error>
```

NaterGator, 2016-07-15:
The new download links are at http://qgroundcontrol.io/downloads/

What page were you linked from?

shadow1runner, 2016-07-15:
Ou, I see.
I took the first google result which was [http://qgroundcontrol.org/downloads](http://qgroundcontrol.org/downloads)

DonLakeFlyer, 2016-07-15:
That page is old. It's not longer connected to anything. But I'll update it anyway since search engines may take a while to catch up.

shadow1runner, 2016-07-16:
Thanks Don,
if the page got completely obsoleted by qgroundcontrol.io maybe a static redirect would help both users and web crawlers.

DonLakeFlyer, 2016-07-19:
I've updated the old page to point to the new one.

closed 2016-07-19 by DonLakeFlyer