mavlink/qgroundcontrol #3807: Plan screen home position not resetting

dagar, 2016-07-16:
This is from doing some testing in HIL. The fly screen properly shows home after it's published by PX4, but plan didn't update.

[image]

DonLakeFlyer, 2016-12-08:
Not sure what you mean here. "Planned home position" doesn't track the vehicle home position automatically.

dagar, 2016-12-09:
In PX4 you get a home position when you first power on (GPS fix), and then it updates anytime you go from LANDED to not. If you restore a mission from file it places the home in the center of the waypoints, but keeps it there even after you get a home update from the vehicle. If you then switch over to the fly view home is in the right position (from the vehicle), but doesn't match plan. There's a message for setting home position, but I don't see that being used.

Did that make sense? I can provide more screenshots/steps otherwise.
I think it's mainly calls to _addPlannedHomePosition that need to be a bit different for PX4. https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/MissionController.cc#L1209

Overlaps with this issue - https://github.com/mavlink/qgroundcontrol/issues/3336

DonLakeFlyer, 2016-12-24:
This is really the same as #3336 which is that the Planned Home Position concept needs a revamp. As it currently stands Planned Home Position is not meant to track the actual Vehicle home position.

closed 2016-12-24 by DonLakeFlyer