mavlink/qgroundcontrol #3809: QGC Daily build / 3.0RC crashes when trying to flash firmware

dagar, 2016-07-16:
https://groups.google.com/forum/#!topic/qgroundcontrol/nz_C1MVqWw4

NaterGator, 2016-07-16:
We don't ship openssl libs like libeay32.dll with QGC. This user has an incompatible build of OpenSSL in their system's path.

Per Qt's [deployment instructions](http://doc.qt.io/qt-5/windows-deployment.html#additional-libraries):
> From Qt version 5.2 onwards, the officially supported version for OpenSSL is 1.0.0 or later. Versions >= 0.9.7 and < 1.0.0 might work, but are not guaranteed to.

dagar, 2016-07-16:
Any reason we shouldn't just add it to the installer?

NaterGator, 2016-07-16:
Nope, assuming we comply with [export](http://doc.qt.io/qt-5/opensslsupport.html) [restrictions](https://en.wikipedia.org/wiki/Export_of_cryptography_from_the_United_States).

DonLakeFlyer, 2016-07-17:
We used to require OpenSSL install, then I pulled out all the ssl links to avoid it. No with pulling github release looks like we need it again. Seems like I remember not being able to use github apis without ssl.

DonLakeFlyer, 2016-07-17:
And whether you can actually add an OpenSLL installer to QGC installer without hitting export restrictions is clear as mud.

NaterGator, 2016-07-17:
Hmm... maybe we can get around it by hosting some dlls on S3 and then having a little dialog that pops up when you first try to flash firmware. IANAL but if it says something to the effect of "you agree to abide by all applicable export restrictions by confirming blah blah blah" and confirming downloads the DLLs in qgroundcontrol's directory maybe that is good enough?

DonLakeFlyer, 2016-07-17:
I asked Bill Bonney about it. APM Planner 2 has OpenSSL included in its installer. I think for 3.0 we doc it for now that you need the separate install. I'm going to take a quick look to see about implementing the SSL support check to at least let the user know they are missing something.

DonLakeFlyer, 2016-07-17:
I"ve updated the user guide to talk about OpenSSL install. Still need to:
* Update main web site windows download link to not go directly to installer, but go to download user guide page instead.
* Look into doing the SSL check in QGC 3.0 to warn using of missing install

DonLakeFlyer, 2016-07-18:
I've update the main windows download link on the website to go to the installer page in the user guide instead of direct link to installer.

DonLakeFlyer, 2016-07-23:
All taken care of

closed 2016-07-23 by DonLakeFlyer