mavlink/qgroundcontrol #3820: Support airspeed sensor calibration on all airframes (feature request)

tstellanova, 2016-07-18:
I would like to attach a pitot airspeed sensor to a multicopter airframe and compare the measured airspeed with the estimated velocity. However, because this is a multicopter airframe, qgroundcontrol doesn't present an option to calibrate the airspeed sensor.  The calibration option is apparently only presented by qgc on airframes for which an airspeed sensor is required (eg VTOL and FW). 

One workaround is to use nsh and the commander command line "commander calibrate airspeed" to calibrate the sensor.  However, it would be great if the presence of the airspeed sensor could be detected by qgc, and the calibration option presented.

DonLakeFlyer, 2016-07-19:
As far as I know there is no way for QGC to detect an airspeed sensor. Need support on the firmware side.

dagar, 2016-10-20:
I think the right way to do this would be SYS_STATUS onboard_control_sensors_present. It's already partially implemented in PX4, we'd just need differential pressure set by the airspeed sensor drivers and maybe a few others while we're at it.

MAV_SYS_STATUS_SENSOR_DIFFERENTIAL_PRESSURE
MAV_SYS_STATUS_SENSOR_GPS
etc

I can implement the PX4 side if someone wants to update QGC sensor calibration.

DonLakeFlyer, 2016-10-21:
A little hard to figure out to handle things that show up all of a sudden. These bits in status don't really work if they aren't implemented immediately as soon as they become available. Otherwise no way to tell if the bit is not set because there is no airspeed. Or if the bit is not set because the firmware doesn't support it yet.