mavlink/qgroundcontrol #3824: Sensor calibration params not shown correctly (2.9.7b)

julianoes, 2016-07-19:
I calibrated the sensors of my Pixracer and I the calibration values don't show up properly in the parameters view, even after a restart or refresh. The sensor calibration is indicated "green" though.

[image]

The params seem to be set if I check in the NuttShell:

```
nsh> param show CAL_ACC0*
Symbols: x = used, + = saved, * = unsaved
x + CAL_ACC0_ID [356,599] : 1442826
x + CAL_ACC0_XOFF [357,600] : 0.1131
x + CAL_ACC0_YOFF [358,601] : 0.1171
x + CAL_ACC0_ZOFF [359,602] : 0.2478
x + CAL_ACC0_XSCALE [360,603] : 0.9969
x + CAL_ACC0_YSCALE [361,604] : 0.9988
x + CAL_ACC0_ZSCALE [362,605] : 0.9933
```

QGroundControl 2.9.7b 24c4e9d, Ubuntu 16.04

DonLakeFlyer, 2016-07-19:
Use a daily build instead or a 3.0 RC.

julianoes, 2016-07-30:
I actually just tested this on 3.0 and it's much clearer, thanks!

closed 2016-07-30 by julianoes